geerlingguy/rpi-clone #4: Feature question

irishmonk-57, 2024-02-10:
OK, I'll confess: TL;DR.  But I have a question:

I know that rpi-clone can write a copy of a "live" OS from an SD card to an NVME card. I used this feature last night to clone my SD-based OS to the new NVME, and happy to report that worked fine - I was able to boot from the NVME. Whoopee! 

My question is this: Can `rpi-clone` write/clone a **_single partition from an image file to an unmounted device_**? IOW, if I had an image file on external storage (LAN, USB stick, etc), an SD card, and an NVME all connected to my RPi5, could `rpi-clone` write the root partition of the image file to an (un-mounted) SD or NVME card? 

And apologies for the TL;DR question... I have read through the README here, but it's rather detailed, and I'm unfamiliar with `rpi-clone` as I only used it once in a "scripted" situation. The balance of this "issue" may be irrelevant. It's simply a description of the situation behind my question, offered as background which might lead the reader to suggest alternative solutions.

### BACKGROUND:
I have a `.img` file created as a backup for a **RPi5** by the program [`image-backup`](https://forums.raspberrypi.com/viewtopic.php?t=332000).  Through no fault of `image-backup`, the latest `.img` file got "buggered up" - **_but_** all of the "buggering" was in the `boot` partition. So - I have an image file with a **good root** partition, but a **buggered boot** partition. 

I had some issues with a recent `apt` upgrade, and the system seemed out of sorts, so I decided to restore my latest backup image. Without knowing that the latest backup image was buggered, I flashed the buggered image file to my SD card **:(**  (lesson learned). And of course the system refused to boot. I didn't have the NVME when this happened, and after some flailing about I decided to re-install using the RPi Imager & the latest "Lite" distro from the rpi.com website. 

In the meantime, I had an opportunity to inspect the buggered image/backup file. All indications are that the root `/` partition is sound, but the `/boot` partition is another story. This is actually a relief because the root partition has been extensively changed since install - whereas the `/boot` partition has not. Anyway - more flailing about ensued in an effort to restore the `root` partition - all unsuccessful to date. 

So that's about it. I hope this is at least intelligible. I'm posting here because of the successful experience w/ rpi-clone last night. Apologies if it's horribly off-topic.

geerlingguy, 2024-02-11:
No problem—IMO the primary mission of `rpi-clone` is to clone a booted Pi drive (all partitions) to another drive. Ideally it could be [any type of drive] to [any other type] (and maybe to/from an ISO at some point?), but I don't plan on adding features like specialized partitioning schemes.

irishmonk-57, 2024-02-11:
Oh -   I wasn't thinking on the lines of specialized partitioning schemes, but RPi does have 2 (or 3?) partitions now, and of course those partitions are all contained as distinct entities in the raw image file. I was wondering if there might be a way in `rpi-clone` to address my situation: 

* one image has a solid root partition, and a buggered boot partition
* other images have solid root and boot partitions. 

Unfortunately for me, the image w/ the buggered boot partition has a lot of work in it - but all the work is in the root partition. If I could "graft" a solid boot partition onto the image with the buggered boot partition, I'd have a whole, bootable system again. 

Anyway, sorry for the bother. It's still early days for you & rpi-clone, and I'm sure you've got other things on your mind.