geerlingguy/rpi-clone #11: —exclude examples

mariomaz87, 2024-03-14:
Hi, I’m reading the documentation but I don’t understand the usage of the exclude command. I want to exclude some folders from a clone, but how can I select them? I’m familiar with rclone —exclude but in this case I would have also to specificy the partition and I don’t know how. Thanks!

fmarzocca, 2024-03-14:
Maybe I didn't understand your problem, but this is how I am dealing with excluding folders:

`rpi-clone sda -u --exclude=/home/john/myfolder`

mariomaz87, 2024-03-14:
> Maybe I didn't understand your problem, but this is how I am dealing with excluding folders:
> 
> `rpi-clone sda -u --exclude=/home/john/myfolder`

Thanks. This answer my problem :)

framps, 2024-03-30:
> This answer my problem :)

Great. But please close this issue now :wink

closed 2024-03-31 by mariomaz87