geerlingguy/rpi-clone #12: Backup failed

Unclerojelio, 2024-03-24:
I attempted to backup the SD card in my Raspberry Pi 4 with the command:

```
# rpi-clone sda
```
I got this in response:

```
Error: /dev/sda: unrecognised disk label

Booted disk: mmcblk0 31.9GB                Destination disk: sda 128.0GB
---------------------------------------------------------------------------
Part               Size    FS     Label           Part   Size  FS  Label  
1 /boot/firmware   512.0M  fat32  --                                      
2 root              29.2G  ext4   rootfs                                  
---------------------------------------------------------------------------
== Initialize: IMAGE partition table - partition number mismatch: 2 -> 0 ==
1 /boot/firmware      (61.5M used)   : MKFS  SYNC to sda1
2 root                (2.3G used)    : RESIZE  MKFS  SYNC to sda2
---------------------------------------------------------------------------
Run setup script       : no.
Verbose mode           : no.
-----------------------:
** WARNING **          : All destination disk sda data will be overwritten!
-----------------------:

Initialize and clone to the destination disk sda?  (yes/no): yes
Optional destination ext type file system label (16 chars max): 

Initializing
  Imaging past partition 1 start.
  => dd if=/dev/mmcblk0 of=/dev/sda bs=1M count=8 ...
  Resizing destination disk last partition ...
    Resize failed.  See /tmp/rpi-clone-output.
    Try running rpi-clone again.

fdisk: cannot open /dev/sda: No such file or directory
```

/tmp/rpi-clone-output contains:

```
sfdisk: cannot open /dev/sda: No such file or directory

====rpi-clone
==orig:
label: dos
label-id: 0x0200edf6
device: /dev/mmcblk0
unit: sectors
sector-size: 512

/dev/mmcblk0p1 : start=        8192, size=     1048576, type=c
/dev/mmcblk0p2 : start=     1056768, size=    61277184, type=83

==edited:
label: dos
label-id: 0x0200edf6
device: /dev/mmcblk0
unit: sectors
sector-size: 512

/dev/mmcblk0p1 : start=        8192, size=     1048576, type=c
/dev/mmcblk0p2 : start=     1056768,  type=83
```

At this point `lsblk` shows only mmcblk mounted. sda does not appear even after disconnecting and reconnecting. Is there a way to recover from this?

matthijskooijman, 2024-04-05:
Are you sure `/dev/sda` exists before you run this script? Looks like it doesn't, and then this would be completely outside of rpi-clone?

Unclerojelio, 2024-04-05:
`lsblk` showed that it existed before I ran the script.

fmarzocca, 2024-04-05:
> `lsblk` showed that it existed before I ran the script.

I would try to insert the backup card in another USB slot. Maybe that one is defective?