geerlingguy/rpi-clone #13: Cloning from running SD card to external card fails - cannot boot from cloned card

carylewis, 2024-03-25:
I recently started using bookworm on my raspberry pi 4b, and since then the rpi-clone utility does not work.

It does not generate any errors, but the system will not boot from a cloned card.

The Pi reports "Firmware not found" on the display.

Using the Pi's sd card clone utility works fine.

Please let me know how I can troubleshoot rpi-clone.

fmarzocca, 2024-03-25:
Did you install latest version 2.0.23?

carylewis, 2024-03-25:
Yes, I did a fresh install before posting.

carylewis, 2024-03-26:
I ran sudo `rpi-clone -x -v -f sdb | tee out, `and didn't see any references to making the the sd card bootable? Is rpi-clone supposed to do that, or is there a manual step I should be performing?

Here is the start of the log:

In```
itializing
  Imaging past partition 1 start.
  => dd if=/dev/mmcblk0 of=/dev/sdb bs=1M count=8 ...
  Resizing destination disk last partition ...
    Resize success.
  Changing destination Disk ID ...
  => mkfs -t vfat -F 32  /dev/sdb1 ...
  => mkfs -t ext4  /dev/sdb2 ...
```

carylewis, 2024-03-27:
If I use piclone to copy the running system to an external card, that card boots fine, and then I can use rpi-clone without the -f option  to update it, it continues to boot correctly.

fmarzocca, 2024-03-27:
> If I use piclone to copy the running system to an external card, that card boots fine, and then I can use rpi-clone without the -f option to update it, it continues to boot correctly.

That's interesting, Some developer should jump in....

fmarzocca, 2024-03-28:
An interesting test could be this:
- initialize the SD with `sudo rpi-clone -f sdb`
- at the end run a simple update with `sudo rpi-clone sdb`
- check if the clone boots

carylewis, 2024-05-30:
> An interesting test could be this:
> 
> * initialize the SD with `sudo rpi-clone -f sdb`
> * at the end run a simple update with `sudo rpi-clone sdb`
> * check if the clone boots

This did not work.

carylewis, 2025-01-23:
Just tried this again on a new sd card, and the cloned card does not boot.

framps, 2025-01-23:
Please read [this](https://github.com/geerlingguy/rpi-clone/issues/44#issuecomment-2554292937) and try my patched code.

framps, 2025-01-24:
@geerlingguy accepted my PR in the meantime. @carylewis So you also can now just use latest code level from this repo instead of the code from my fork.