geerlingguy/rpi-clone #14: rsync_xal_set: lsetxattr("/mnt/clone/boot","btrfs.compression") failed: Operation not supported (95)

b-morgan, 2024-03-30:
I'm not sure if this is something that can be fixed. I'm running rpi-clone on a Le Potato AML-S905X-CC with the OS installed on an eMMC. Attached is a transcript of the session.

[typescript.zip](https://github.com/geerlingguy/rpi-clone/files/14812742/typescript.zip)

matthijskooijman, 2024-04-05:
Somehow some extended attribute is not supported while cloning. It looks like the filesystem types match up (even the rootfs is btrfs and sdb2 is ext before cloning, sdb2 is initialized to btrfs because of the fstype mismatch), so I suspect there might be mount options needed that are present on the rootfs, and not on the temporary mount.

Can you paste the output of `mount` on this system?

b-morgan, 2024-04-05:
Here is the output you requested (with an `lsblk` as well). The system is booted off of the 32GB eMMC 
 in mmcblk0 and there is a 16GB microSD card in mmcblk1.
```
li@libre:~ $ lsblk
NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0      179:0    0 29.1G  0 disk
├─mmcblk0p1  179:1    0  256M  0 part /boot/efi
└─mmcblk0p2  179:2    0 28.9G  0 part /
mmcblk0boot0 179:32   0    4M  1 disk
mmcblk0boot1 179:64   0    4M  1 disk
mmcblk1      179:96   0 14.8G  0 disk
├─mmcblk1p1  179:97   0    1G  0 part /media/li/opi_boot
└─mmcblk1p2  179:98   0  2.7G  0 part /media/li/opi_root
li@libre:~ $ mount
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
udev on /dev type devtmpfs (rw,nosuid,relatime,size=846124k,nr_inodes=211531,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,noexec,relatime,size=197592k,mode=755)
/dev/mmcblk0p2 on / type btrfs (rw,noatime,ssd,space_cache,subvolid=5,subvol=/)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate,memory_recursiveprot)
efivarfs on /sys/firmware/efi/efivars type efivarfs (ro,nosuid,nodev,noexec,relatime)
bpf on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,relatime,mode=700)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=29,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
mqueue on /dev/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime)
sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
hugetlbfs on /dev/hugepages type hugetlbfs (rw,relatime,pagesize=2M)
debugfs on /sys/kernel/debug type debugfs (rw,nosuid,nodev,noexec,relatime)
fusectl on /sys/fs/fuse/connections type fusectl (rw,nosuid,nodev,noexec,relatime)
configfs on /sys/kernel/config type configfs (rw,nosuid,nodev,noexec,relatime)
/dev/mmcblk0p1 on /boot/efi type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=197592k,nr_inodes=49398,mode=700,uid=1000,gid=1000)
gvfsd-fuse on /run/user/1000/gvfs type fuse.gvfsd-fuse (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
/dev/mmcblk1p1 on /media/li/opi_boot type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
/dev/mmcblk1p2 on /media/li/opi_root type ext4 (rw,nosuid,nodev,relatime,errors=remount-ro,uhelper=udisks2)
```

matthijskooijman, 2024-04-05:
This is the most relevant line:

```
/dev/mmcblk0p2 on / type btrfs (rw,noatime,ssd,space_cache,subvolid=5,subvol=/)
```

I was expecting something like an `xattr` option in there, but nothing here looks like that (but I also have no experience with btrfs, so maybe one of these options is relevant).

Maybe you could try manually mounting `/dev/sdb2` somewhere and then seeing what `mount` outputs about that mount, see if there's a difference in mount options?