geerlingguy/rpi-clone #15: Backup failed though procedure looked fine

scargill, 2024-04-01:
Hi 
I've been using Bills rpi-clone for years, backing up my rpi 4 daily.
I recently (Feb) bought an RPI5 and installed the latest RPI 64 bit OS from the RPI site. Weeks ago we cloned the SD to a USB-mounted SSD - worked fine - checked it - no problem. Done this hundreds of times. Been backing up ever since but not checked in last 4 weeks.  When I tried to make a clone today and run it, the clone would not work (even tried clean clone using the fdisk option).

So I deleted rpi-clone and grabbed yours - and a fresh PI OS 64 bit - ran sudeo apt update and sudo apt upgrade - exactly the same. Not working - and I've tried 2 different SSDs (the ones I normally use) - made no difference - so it looks like a recent - past days or weeks  has not only killed BILL's  rpi-clone but yours too. As Bill has not updated rpi-clone in 4 years I'm counting on you. 

Looks absolutely fine when creating teh clone - but it will not boot.

HELP.

framps, 2024-04-01:
> but it will not boot.

Most of the time that's because the configured PARTUUIDs don't match the existing PARTUUIDs. Actually rpi-clone will update them. But I suggest first to invoke [this script](https://github.com/framps/raspberryTools/blob/master/syncUUIDs.sh) which reports the configured and existing PARTUUIDs. 

If they match - do you have a monitor connected and can show the boot messages?

scargill, 2024-04-01:
```

pi@rpi5:~ $ sudo bash syncUUIDs.sh /dev/sda
syncUUIDs.sh 0.1.1 (https://github.com/framps/raspberryTools)
--- Root PARTUUID ac2d0382-02 already used in /dev/sda1/cmdline.txt
--- Boot PARTUUID ac2d0382-01 already used in /dev/sda2/etc/fstab
--- Root PARTUUID ac2d0382-02 already used in /dev/sda2/etc/fstab
pi@rpi5:~ $
```
3 flashing blue on SD then nothing - some time later another 3...  still not booting...
I don;t have a monitor - always used ssh with RPis - if I go back to SD - are there 2 files I can compare by mounting the SSD and looking inside?

scargill, 2024-04-02:
Open to ideas - I utterly rely on rpi-clone and have done for years to get me out of a jam - and right now, without it my RPI05 is sitting on the shelf waiting for a solution.

framps, 2024-04-02:
The easiest way to make progress would be to connect a monitor and provide a screenshot of the boot messages. Don't you have any old spare monitor handy? I frankly suggest to everybody who uses a headless system to have a monitor handy - just for these kind of boot issues. I just bought a tiny 10'' LCD monitor exactly for this. I was bored to move my big 17'' monitor I use on my desktop downstairs whenever I had boot issues with my Raspberries located in the basement. 

Next you can modify the OS to make the systemd log persistent (I assume you use Bookworm). Then create a clone and  inspect the persistent systemd log on another Linux box. I frankly never didn't go this path and only know from theory this should work.

scargill, 2024-04-02:
Out now.. Going to follow this up. Can get to a monitor....   
  

  

  
  
Regards
  

  
Pete
  
  

  
  
  
Peter Scargill
  
https://tech.scargill.net
  
Calle Ceuta 14
  
Galera
  
Granada 18840
  
Spain
  
+34 634 326331

matthijskooijman, 2024-04-05:
If you have a (3.3V) USB-to-TTL-serial adapter (or another rpi could be used) then it might also be useful to log the serial console - IIRC the bootloader outputs more info there on the early boot process.

scargill, 2024-04-07:
> The easiest way to make progress would be to connect a monitor and provide a screenshot of the boot messages. Don't you have any old spare monitor handy? I frankly suggest to everybody who uses a [...]

You;r going to love this - message 1 
[image]

message 2
[image]

Am I right in saying it maybe sees the SSD - but not enough power? I've no idea how to get to the config.txt on a Windows PC with that SSD ??

And i am using the proper RPi5 power supply (supposedly 4A at 5v. Nothing else connected to the RPi5 - just the SSD in an adaptor - and the monitor... should I trust that message - 5A seems awfully high...