geerlingguy/rpi-clone #29: Doesn't work in Manjaro Linux...

Pti-Jean, 2024-05-13:
(Version translated into English):
Hello,

rpi-clone does not work on AstroArch:
https://github.com/devDucks/astroarch
which is based on Manjaro Linux...
It stops on:
` => rsync /boot/ /mnt/clone/boot ...`
and that's as far as it goes...
Should we consider a new version?

####
(Original version in French):
Bonjour,

rpi-clone ne fonctionne pas sur AstroArch:
https://github.com/devDucks/astroarch
qui est basé sur Manjaro Linux...
Cela s'arrête sur:
`  => rsync /boot/ /mnt/clone/boot  ...`
et cela ne va pas plus loin...
Faudrait-il envisager une nouvelle version ?

JM

Pti-Jean, 2024-05-15:
(Version translated into English):
Despite this bug, the copy took place, only the configuration of the /boot/cmdline.txt and /etc/fstab files did not...
So I looked at the UUIDs of the partitions with the command:
`ls -l /dev/disk/by-uuid/`
and I transferred these UUIDs to the 2 configuration files mentioned above, and then the clone worked!
The bug shouldn't be too hard to fix! No ?

(Original version in French):
Malgré ce bug, la copie a bien eu lieu, seule la configuration des fichiers /boot/cmdline.txt et /etc/fstab n'a pas eu lieu...
J'ai donc regardé les UUID des partitions avec la commande:
`ls -l /dev/disk/by-uuid/`
et j'ai reporté ces UUID dans les 2 fichiers de configuration cité ci-dessus, et après le clone a fonctionné!
Le bug ne devrait pas être très compliqué à corrigé! Non ?
JM

Pti-Jean, 2024-08-11:
(Version translated into English):
rpi-clone plants on this line:
`$ ps -ef | grep grep`
`root        2037    1717  0 15:23 pts/3    00:00:00 grep -q /mnt/clone/boot/cmdline.txt`

(Original version in French):
rpi-clone plante sur cette ligne:
`$ ps -ef | grep grep`
`root        2037    1717  0 15:23 pts/3    00:00:00 grep -q /mnt/clone/boot/cmdline.txt`

JM

Pti-Jean, 2025-11-22:
(Version translated into English):
Hello,

I think this is a long and short UUID issue... because the system has been copied, I just need to put the correct UUIDs back in the /boot/cmdline.txt and /etc/fstab files for it to work...

FYI:
`$ ls -l /dev/disk/by-uuid/
total 0
lrwxrwxrwx 1 root root 10 22 nov.  17:39 9db611c4-9be5-4854-8ac2-8b3915cf9b6c -> ../../sda2
lrwxrwxrwx 1 root root 10 22 nov.  17:39 C365-B8A0 -> ../../sda1
$ cat /boot/cmdline.txt 
root=UUID=9db611c4-9be5-4854-8ac2-8b3915cf9b6c rw rootwait console=serial0,115200 console=tty1 fsck.repair=yes
$ cat /etc/fstab 
\# Static information about the filesystems.
\# See fstab(5) for details.

\# \<file system> \<dir> \<type> \<options> \<dump> \<pass>
UUID=C365-B8A0  /boot   vfat    defaults        0       0
UUID=9db611c4-9be5-4854-8ac2-8b3915cf9b6c / ext4 defaults 0 1`

JM

(Original version in French):
Bonjour,

Je pense qu'il s'agit d'un problème d'UUID long et court... car le système a été copié, il me faut juste remettre les bons UUID dans les fichiers /boot/cmdline.txt et /etc/fstab pour que cela fonctionne...

Pour info:
`$ ls -l /dev/disk/by-uuid/
total 0
lrwxrwxrwx 1 root root 10 22 nov.  17:39 9db611c4-9be5-4854-8ac2-8b3915cf9b6c -> ../../sda2
lrwxrwxrwx 1 root root 10 22 nov.  17:39 C365-B8A0 -> ../../sda1
$ cat /boot/cmdline.txt 
root=UUID=9db611c4-9be5-4854-8ac2-8b3915cf9b6c rw rootwait console=serial0,115200 console=tty1 fsck.repair=yes
$ cat /etc/fstab 
\# Static information about the filesystems.
\# See fstab(5) for details.

\# \<file system> \<dir> \<type> \<options> \<dump> \<pass>
UUID=C365-B8A0  /boot   vfat    defaults        0       0
UUID=9db611c4-9be5-4854-8ac2-8b3915cf9b6c / ext4 defaults 0 1`

JM

framps, 2025-11-23:
I forgot syncUUIDs checks for RaspbainOS. Sorry. 

If rpi-clone also fails at the same place when you use PARTUUIDs that's strange.

Please show all messages you get when you invoke rpi-clone on your system using PARTUUIDs.

Pti-Jean, 2025-11-23:
`$ sudo fdisk -l /dev/sdb
Disque /dev/sdb : 931,51 GiB, 1000204886016 octets, 1953525168 secteurs
Modèle de disque : External USB 3.0
Unités : secteur de 1 × 512 = 512 octets
Taille de secteur (logique / physique) : 512 octets / 4096 octets
taille d'E/S (minimale / optimale) : 4096 octets / 4096 octets
Type d'étiquette de disque : dos
Identifiant de disque : 0x768b4ec4

Périphérique Amorçage   Début        Fin   Secteurs Taille Id Type
/dev/sdb1                2048    1050623    1048576   512M  c W95 FAT32 (LBA)
/dev/sdb2             1050624 1953525167 1952474544   931G 83 Linux`

JM

framps, 2025-11-23:
Ah ... yes, that's a good idea. Just change the locale for the current session you want to run rpi-clone 👍 That way you can keep your French locale.