geerlingguy/rpi-clone #33: clone to an img file

aaron-gh, 2024-06-27:
It would be incredibly nice for cm4 systems if we could backup to an img file for reimaging later.

These images could then be taken by the user and placed where ever they like, on a remote server etc.

framps, 2024-06-27:
It's possible with a [small update in rpi-clone](https://github.com/framps/rpi-clone/tree/m_33)  :wink: 

1) Allocate a file which is large enough to get the clone, i.e. `truncate -s 10G /backup/clone.dd` for a 8GB image
2) Mount the file into a loop device with `sudo losetup -f /backup/clone.dd`
3) Invoke rpi-cone with `sudo rpi-clone loop0`
4) Release loop device with `sudo losetup -d loop0`

Done :wink:

aaron-gh, 2024-06-30:
Hi,
I'm having some issues with this approach.
At the point where rpi-clone tries to set up the filesystem on loop0 it fails to read the partition table, and then 
says that there is no partition on /dev/loop02.
Is there something I'm missing?
Thanks.

framps, 2024-06-30:
> Is there something I'm missing?

I'm sorry. When I updated the original rpi-clone in my github clone I had [a typo](https://github.com/framps/rpi-clone/commit/5a3e7838b38cebd79521bbb5b7fd8a3aca10e66a) :disappointed: which I just fixed. Please grab the updated code :wink: 

```
pi@raspberrypi:~ $ sudo ./rpi-clone -f loop0
Error: /dev/loop0: unrecognised disk label

Booted disk: mmcblk0 31.3GB                Destination disk: loop0 17.2GB
---------------------------------------------------------------------------
Part               Size    FS     Label           Part   Size  FS  Label  
1 /boot/firmware   512.0M  fat32  --                                      
2 root              14.8G  ext4   rootfs                                  
---------------------------------------------------------------------------
== Initialize: IMAGE partition table - forced by option ==
1 /boot/firmware      (64.0M used)   : MKFS  SYNC to loop0p1
2 root                (10.3G used)   : RESIZE  MKFS  SYNC to loop0p2
---------------------------------------------------------------------------
Run setup script       : no.
Verbose mode           : no.
-----------------------:
** WARNING **          : All destination disk loop0 data will be overwritten!
-----------------------:

Initialize and clone to the destination disk loop0?  (yes/no): yes
Optional destination ext type file system label (16 chars max): 

Initializing
  Imaging past partition 1 start.
  => dd if=/dev/mmcblk0 of=/dev/loop0 bs=1M count=8 ...
  Resizing destination disk last partition ...
    Resize success.
  Changing destination Disk ID ...Re-reading the partition table failed.: Invalid argument

  => mkfs -t vfat -F 32  /dev/loop0p1 ...
  => mkfs -t ext4  /dev/loop0p2 ...

Syncing file systems (can take a long time)
Syncing mounted partitions:
  Mounting /dev/loop0p2 on /mnt/clone
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
  => rsync // /mnt/clone with-root-excludes ...
  Mounting /dev/loop0p1 on /mnt/clone/boot/firmware
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
  => rsync /boot/firmware/ /mnt/clone/boot/firmware  ...

Editing /mnt/clone/boot/firmware/cmdline.txt PARTUUID to use d88a0cbe
Editing /mnt/clone/etc/fstab PARTUUID to use d88a0cbe
===============================
Done with clone to /dev/loop0
   Start - 21:13:34    End - 21:16:51    Elapsed Time - 3:17

Cloned partitions are mounted on /mnt/clone for inspection or customizing.
```
**Please note:** There are still some messages which require additional updates in rpi-clone to vanish. In addition the backup is not shrinked. This requires additional code in rpi-clone. Not difficult to implement but requires some development time. Maybe somebody will grab code, make the code robust, add code to shrink the image and finally create a PR and add the backup creation feature in rpi-clone.

framps, 2024-08-09:
@aaron-gh Any update on this issue?