geerlingguy/rpi-clone #37: rpiclone fails w same error

anigupta, 2024-07-28:
Hello,

Apologies if this has already been reported, but I didn't find a similar issue.

Been trying rpi-clone on a new Pi5 with a 1tb Kingspec NVMe memory controller, using a waveshare (broadcom chip) M.2 hat. The NVMe works just fine. Have been playing with partitions and transferring data etc. Waveshare hat also is installed correctly. Can anyone guide me on this error? Any help is appreciated. Thank you in advance. 

This is the error I get everytime:

```❯ sudo rpi-clone nvme0n1

Booted disk: mmcblk0 63.9GB                Destination disk: nvme0n1 1.0TB
---------------------------------------------------------------------------
Part               Size    FS     Label           Part   Size    FS     Label
1 /boot/firmware   512.0M  fat32  --              1      200.0M  fat32  --
2 root              59.0G  ext4   rootfs          2      115.5G  --     --
                                                  3      837.9G  --     --
---------------------------------------------------------------------------
== Initialize: IMAGE partition table - FS types conflict ==
1 /boot/firmware      (64.0M used)   : MKFS  SYNC to nvme0n1p1
2 root                (2.7G used)    : RESIZE  MKFS  SYNC to nvme0n1p2
---------------------------------------------------------------------------
Run setup script       : no.
Verbose mode           : no.
-----------------------:
** WARNING **          : All destination disk nvme0n1 data will be overwritten!
-----------------------:

Initialize and clone to the destination disk nvme0n1?  (yes/no): yes
Optional destination ext type file system label (16 chars max):

Initializing
  Imaging past partition 1 start.
  => dd if=/dev/mmcblk0 of=/dev/nvme0n1 bs=1M count=8 ...
  Resizing destination disk last partition ...
    Resize success.
  Changing destination Disk ID ...
  => mkfs -t vfat -F 32  /dev/nvme0n1p1 ...
  => mkfs -t ext4  /dev/nvme0n1p2 ...

Syncing file systems (can take a long time)
Syncing mounted partitions:
  Mounting /dev/nvme0n1p2 on /mnt/clone
mount: /mnt/clone: wrong fs type, bad option, bad superblock on /dev/nvme0n1p2, missing codepage or helper program, or other error.
       dmesg(1) may have more information after failed mount system call.
    Mount failure of /dev/nvme0n1p2 on /mnt/clone.
Aborting!```

Here's the hardware info
```❯ sudo lspci
0000:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 21)
0000:01:00.0 Non-Volatile memory controller: MAXIO Technology (Hangzhou) Ltd. NVMe SSD Controller MAP1202 (rev 01)
0001:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 21)
0001:01:00.0 Ethernet controller: Raspberry Pi Ltd RP1 PCIe 2.0 South Bridge```

framps, 2024-07-28:
I don't own a RPi5 but a CM4 which uses also NVMe storage and I was able to reproduce your issue.

rpi-clone misses a logging facility - unfortunately :-(( 

I'll check if I find the root cause. Unfortunately the rpi-clone code is very tricky :-((

framps, 2024-07-29:
It's not an issue with nvme. It looks like your issue is related to the fact the discovery whether you already have an existing clone on your backup device fails for some reason.

Try `sudo ./rpi-clone -f nvnme0n1`

anigupta, 2024-07-30:
Thanks for the response @framps . There was no data on the nvme controller. It is brand new. Didn't work with the force option either. In any case, I did a fresh flash install on the nvme drive on my mac and then setup the boot order to boot rpi from it. Its working fine now. I guess the rpi-clone script may have issues. As you mentioned, it is quite tricky, esp without any logging facility.

CNurserDE, 2024-07-30:
Reverting c41720 seemed to fix a similar issue for me

framps, 2024-07-30:
@anigupta Can you confirm reverting to last commit fixes your issue? I created the fix with #28 and frankly I don't understand why this fix causes your issue :thinking: 

If you can confirm the previous version fixes your issue we should revert this fix. This will cause an issue with labels which have spaces but that's just an unlikely edge case.

geerlingguy, 2024-07-30:
@framps if indeed that fixes it, ping me if you need and I'll fast track reverting the PR.

anigupta, 2024-07-30:
> @anigupta Can you confirm reverting to last commit fixes your issue? I created the fix with #28 and frankly I don't understand why this fix causes your issue 🤔
> 
> If you can confirm the previous version fixes your issue we should revert this fix. This will cause an issue with labels which have spaces but that's just an unlikely edge case.

Unfortunately, the NVMe is now the boot drive for the Rpi, which is in use, and I don't have any spare NVMe to test it on.

framps, 2024-07-30:
> Unfortunately, the NVMe is now the boot drive for the Rpi, which is in use, and I don't have any spare NVMe to test it on.

I frankly expected this :cry: 

rpi-clone doesn't miss a log but it also misses runtime error capturing. I added runtime capturing in my [debug branch](https://github.com/framps/rpi-clone/tree/debug) but then detected there are a lot of places where rpi-clone handles runtime errors further on but if you enable runtime error handling with `set -euo pipefail ` it's a mess to fix the expected runtime failures  :cry:.

Maybe somebody is willing to use my debug code which catches any runtime error and update rpi-clone accordingly.

ConnorDY, 2024-07-30:
I was having the same issue described above.

I can also confirm that reverting PR #28 fixed the issue for me.

geerlingguy, 2024-07-30:
I've reverted #28 and hopefully this will solve the problem—I'll tag a new release, and if someone wants to work on a fix for #27 that doesn't break the normal use cases, we can look at merging it in again later. Thanks!

closed 2024-07-30 by geerlingguy