geerlingguy/rpi-clone #38: final error: sed: couldn 't open temporary file/MNT/clone/boot/firmware/sedRnsJN4: Read - only file system

hanyuanzhong, 2024-07-29:
runing sudo rpi-clone sdb：
error：
[image]

fjpdevries, 2024-07-29:
Ik ben tot en met vrijdag 16 augustus 2024 afwezig en zal mijn email niet of beperkt kunnen lezen en beantwoorden. Neem voor urgente zaken contact op met ons kantoor, email ***@***.***, tel. 0224-297819.

Met vriendelijke groet,
FSE Turnstiles b.v.
Jos de Vries

fmarzocca, 2024-07-29:
Do you have usb automount enabled? Disable it.

fmarzocca, 2024-07-29:
see: https://github.com/geerlingguy/rpi-clone/issues/31#issuecomment-2191939181

hanyuanzhong, 2024-07-29:
> see: [#31 (comment)](https://github.com/geerlingguy/rpi-clone/issues/31#issuecomment-2191939181)

Thank you very much for your reply. I just came into contact with rpi-clone. I'm a total newbie/noob at this., I don't know how to disable automount and manually mount the partitions

hanyuanzhong, 2024-07-29:
> 您是否启用了 USB 自动挂载？请将其禁用。

Hello, I tried to use sudo systemctl stop udisks2 to disable automatic mounting, but an error was still reported

fmarzocca, 2024-07-29:
Check if you have `devmon` running and stop it.
Then unmount all usb and run `rpi-clone -f `again.
rpi-clone takes care of mounting what it needs.

hanyuanzhong, 2024-07-29:
> Then unmount all usb

Running "lsblk " shows the following
[image]

Run" ps aux | grep devmon" as shown below
[image]

I don't think I started the auto mount,

sudo rpi-clone -f sdb
[image]
[image]

fmarzocca, 2024-07-29:
what's the output of:
`mount | egrep "mmcblk[0-9]|sd[a-z]"`

hanyuanzhong, 2024-07-29:
> mount | egrep "mmcblk[0-9]|sd[a-z]"

[image]

fmarzocca, 2024-07-29:
your partition /boot/firmware is not mounted? Bad.

Reboot the raspberry, stop automount, unmount any usb and retry.

hanyuanzhong, 2024-07-29:
reboot  the raspberry
[image]
What am I going to run next
1：sudo umount /media/pi/bootfs
2：sudo umount /media/pi/rootfs
3：sudo systemctl stop udisks2
4：sudo systemctl daemon-reload
5：sudo rpi-clone -f sdb
These five orders？

hanyuanzhong, 2024-07-29:
> reboot the raspberry [...]

When the sudo rpi-clone -f sdb command is executed, the lsblk command is displayed
[image]