geerlingguy/rpi-clone #40: Error cloning to /dev/nvme0n1p1

kf6spf, 2024-08-08:
I've been using rip-clone for years. Great software!

I've recently been updating my devices to use NVMe M.2 drives. This works fine on Pi 4b, which sees the drive as /dev/sdb, but on the Pi 5, it sees the drive as /dev/nvme0n1, which seems to confuse the partitioning segment. It wants to make /dev/nvme0n11 not /dev/nvme0n1p1 and the same for p2 (wants n12).

I haven't find a way around this yet.

Is there a workaround or is a patch the the software needed?

framps, 2024-08-09:
> s there a workaround or is a patch the the software needed?

Do you use the current version of rpi-clone from **this** repo?

kf6spf, 2024-08-09:
Ah. I've been using the original tree. Upgraded and NVMe works fine. Thanks!

Michael

----------------------------------------------------------------------
Michael Wallis     kf6spf     (408) 396-9037      ***@***.***

framps, 2024-08-09:
Great it works now. Please close this issue now :wink:

closed 2024-08-09 by kf6spf