geerlingguy/rpi-clone #41: Question for Jeff

framps, 2024-08-09:
I tried to contact you via eMail but didn't get any response :disappointed:  So I now use this way to contact you :smiley: 

I noticed you have his trailer in every issue. As you know I maintain raspiBackup and would like to have this trailer in my repo also. Would be great if you let me know how you managed to get this issue trailer displayed.

[image]

geerlingguy, 2024-08-11:
Oh, I'm assuming that'll show up if you have a `.github/FUNDING.yml` file in your repo, and also have "Sponsorships" checked in the repository settings. I thought it just added a little 'Sponsor' button on the repo home page!

Sorry about the email—I honestly treat it like a firehose, and I dip in when I can, but a lot of emails kind of just go past. Feel free to @ me in an issue here or sometimes even Twitter can be a more immediate channel :D

pmarf, 2024-08-11:
> Oh, I'm assuming that'll show up if you have a `.github/FUNDING.yml` file in your repo, and also have "Sponsorships" checked in the repository settings. I thought it just added a little 'Sponsor' button on the repo home page!

That's all done and the sponsorship button is visible. But there is nothing visible in the issues :-( 

But reading your reply I found the root cause why it's not displayed: If I open the issue as framps it's not there. But if another github user opens the issue it's displayed :-D

> Feel free to @ me in an issue here or sometimes even Twitter can be a more immediate channel :D

Will do. Thank you very much.

framps, 2024-08-11:
> Oh, I'm assuming that'll show up if you have a `.github/FUNDING.yml` file in your repo, and also have "Sponsorships" checked in the repository settings. I thought it just added a little 'Sponsor' button on the repo home page!

That's all done and the sponsorship button is visible. But there is nothing visible in the issues :-( 

But reading your reply I found the root cause why it's not displayed: If I open the issue as framps it's not there. But if another github user opens the issue it's displayed :-D

> Feel free to @ me in an issue here or sometimes even Twitter can be a more immediate channel :D

Will do. Thank you very much.

closed 2024-08-11 by framps