geerlingguy/rpi-clone #42: Cannot clone image

jpkelly, 2024-09-05:
Not sure if it is a bug but when cloning my Pi image
Pi 4b running "Raspbian GNU/Linux 12 (bookworm)"
it fails with the following messages.

```
$ sudo rpi-clone -v sda

  Destination disk partition /dev/sda1 is mounted on /media/pi/bootfs.
  The clone cannot proceed unless it is unmounted.
Do you want to unmount /media/pi/bootfs?  (yes/no): yes

Booted disk: mmcblk0 7.9GB                 Destination disk: sda 7.9GB
---------------------------------------------------------------------------
Part               Size    FS     Label           Part   Size    FS     Label  
1 /boot/firmware   512.0M  fat32  --              1      512.0M  fat32  --     
2 root               6.9G  ext4   rootfs          2        6.9G  --     --     
---------------------------------------------------------------------------
== Initialize: IMAGE partition table - FS types conflict ==
1 /boot/firmware      (115.5M used)  : MKFS  SYNC to sda1
2 root                (5.1G used)    : RESIZE  MKFS  SYNC to sda2
---------------------------------------------------------------------------
Run setup script       : no.
Verbose mode           : yes.
-----------------------:
** WARNING **          : All destination disk sda data will be overwritten!
-----------------------:

Initialize and clone to the destination disk sda?  (yes/no): yes
Optional destination ext type file system label (16 chars max): 

Initializing
  Imaging past partition 1 start.
  => dd if=/dev/mmcblk0 of=/dev/sda bs=1M count=8 ...
  Resizing destination disk last partition ...
    Resize success.
  Changing destination Disk ID ...
  => mkfs -t vfat -F 32  /dev/sda1 ...
  => mkfs -t ext4  /dev/sda2 ...

Syncing file systems (can take a long time)
Syncing mounted partitions:
  Mounting /dev/sda2 on /mnt/clone
mount: /mnt/clone: wrong fs type, bad option, bad superblock on /dev/sda2, missing codepage or helper program, or other error.
       dmesg(1) may have more information after failed mount system call.
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
    Mount failure of /dev/sda2 on /mnt/clone.
Aborting!
```

framps, 2024-09-05:
Would you please provide the output of `lsblk -o +fstype,uuid,partuuid` ?

jpkelly, 2024-09-05:
```
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS    FSTYPE UUID                                 PARTUUID
sda           8:0    1  7.4G  0 disk                                                            
├─sda1        8:1    1  512M  0 part                vfat   AAE0-77E1                            c7dc0faf-01
└─sda2        8:2    1  6.9G  0 part                                                            c7dc0faf-02
mmcblk0     179:0    0  7.4G  0 disk                                                            
├─mmcblk0p1 179:1    0  512M  0 part /boot/firmware vfat   AAE0-77E1                            6c028086-01
└─mmcblk0p2 179:2    0  6.9G  0 part /              ext4   ca2be9ef-61a4-4400-a5a1-dc17d7ea790d 6c028086-02
```

framps, 2024-09-05:
/dev/sda2 was not formatted correctly :-(

jpkelly, 2024-09-05:
I'm thinking this issue was due to a faulty (brand new shiny "industrial") SD card. I tried an identical card and was able to clone successfully.

Thanks for the lsblk command/flags though, very helpful!

fmarzocca, 2024-09-06:
Try this:
`sudo rpi-clone -f sda`

jpkelly, 2024-09-06:
I am finding that initializing the disk via rpi-clone often creates an unbootable image. Subsequently syncing with api-clone makes it bootable.

fmarzocca, 2024-09-06:
> I am finding that initializing the disk via rpi-clone often creates an unbootable image. Subsequently syncing with api-clone makes it bootable.

Yes, this is due to automounting. See [here](https://github.com/geerlingguy/rpi-clone/issues/31#issuecomment-2191939181)