geerlingguy/rpi-clone #43: Ubuntu Server 24.04 - impossible to boot

thehedgefrog, 2024-09-12:
I have an SD card I wrote with the Pi imager of Ubuntu Server 24.04.  When I run `sudo rpi-clone -f sda` (with or without the `-f` option or in succession), the copy succeeds but the USB disk then never boots.

However, if I use the imager to write to the disk directly, it works fine.  I have replicated this on 2 Pis and validated that the files are identical on both the card and the drive.

Am I missing an obvious step?

thehedgefrog, 2024-09-12:
After some digging, it appears the Raspberry Pi imager's image uses labels in fstab and cmdline.txt for Ubuntu, while it uses PARTUUID for Raspbian.  This broke the behavior.

kpanoulas2, 2024-09-12:
I had to tweak the process a bit but now it worked for me. 
I ran sudo rpi-clone -f nvme0n1
When rpi-clone asks for volume name, the name *has* to be: writable
Then before completing the script, the the vfat volume label has to be changed to: system-boot. 
[to match the contents of the /etc/fstab in the new disk] This can be done with the following command:
sudo fatlabel /dev/nvme0n1p1 system-boot
After that the system booted as expected.

fubits1, 2024-09-30:
@kpanoulas2 thank you so much for sharing - this solved the problem!

logoff, 2025-01-30:
Thanks, @kpanoulas2, these steps were crucial. However, after applying these changes, I get another error:

> Failed to open partition 63

It seems the disk is properly detected, but the OS won't boot. Any idea?

framps, 2025-01-30:
Ubuntu has some differences in the partition naming (PARTUUID vs LABEL). Looks like that's the root cause and you have to use @kpanoulas2 workaround.

I never used rpi-clone on Ubuntu. I will do this when I'm back home to understand what's the root cause the workaround has to be used. Maybe I can create a fix.

@logoff Looks like you have some special setup on your system which is different than the standard Ubuntu setup.
Please provide the output of  `lsblk -fs -o +PARTUUID` and `cat /etc/fstab`

logoff, 2025-01-31:
I ended up solving the issue applying @kpanoulas2 steps, plus manually copying files to the boot partition, mounted on `/boot/firmware`. I would have taken more notes of the process, because now I don't remember if I applied more things, and I tried many things 😓.

framps, 2025-02-07:
I can confirm rpi-clone does not support to clone an Ubuntu image

```framp@raspberry-ubuntu-24:/mnt/clone$ sudo blkid | grep -E "(mmcblk|sdc)"
/dev/mmcblk0p1: LABEL_FATBOOT="system-boot" LABEL="system-boot" UUID="105A-0247" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="54921c2b-01"
/dev/mmcblk0p2: LABEL="writable" UUID="1224dcf6-fbfb-4218-b799-6054acb76356" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="54921c2b-02"
/dev/sdc2: UUID="0728249d-822c-422e-8960-a545120234dc" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="713802e9-02"
/dev/sdc1: UUID="AF7C-41AE" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="713802e9-01"
```
/dev/mmcblk is the running Ubuntu and /dev/sdc is the cloned system. Given the labels are not cloned the cloned system will not boot because Ubutuntu uses labels in /etc/fstab and /boot/cmdline.txt :(

**Workaround**:

Execute following commands:
`sudo blkid`
Write down the LABEL of the boot and root partition
`sudo dosfslabel /dev/<cloneDevice>1 system-boot`
`sudo e2label /dev/<cloneDevice>2 writable`

on the device you used to clone your system.

Then your cloned Ubuntu will boot.

Not sure I find some spare time to add Ubuntu support in rpi-clone.

framps, 2025-02-23:
Maybe #16 will fix this issue. Please try the code in the PR.