geerlingguy/rpi-clone #44: Raspberry Pi 5 OS - Impossible to boot

kikaragyozov, 2024-09-18:
I'm trying to ditch my current SD card and switch entirely to a portable SSD connected via USB.

Things I did:

1. Deleted all partitions on my SSD.
2. Manually created partitions equal to the ones on my SD card, with minimum space matching the ones on the SD card:
```
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS           FSTYPE UUID                                 PARTUUID                             LABEL
sda           8:0    0 931.5G  0 disk
├─sda1        8:1    0   512M  0 part /media/pi/FF61-CD85   vfat   FF61-CD85                            deb0b539-01
└─sda2        8:2    0   931G  0 part /media/pi/Extreme Pro ext4   335884ab-a981-4e26-a064-d61879180b66 deb0b539-02                          Extreme Pro
mmcblk0     179:0    0  14.9G  0 disk
├─mmcblk0p1 179:1    0   512M  0 part /boot/firmware        vfat   D735-E67D                            41d867f7-01                          bootfs
└─mmcblk0p2 179:2    0  14.4G  0 part /                     ext4   b69628b0-b5d6-474c-9bed-477305ae01ad 41d867f7-02                          rootfs
```
3. Ran ```sudo rpi-clone sda```
4. Process completed successfully.
5. Used ```raspi-config``` to change boot order to USB.
6. Realized I should have done this before cloning/syncing.
7. Restarted device.
8. Ran again ```sudo rpi-clone sda``` without literally doing anything else.
9. Process completed successfully.
10. Shut down Raspberry PI.
11. Setup failed to boot into SSD, and as a fallback it booted into my SD card.
12. Verified via an HDMI cable connection that the issue for not booting into the SSD (with an unplugged SD card) was due to a ```unable to read partition as FAT (partition 0)``` error.

I'm at a complete loss. Any help would be appreciated. I verified fstab entries - they seem to be correct. PARTUUIDS match with the ones specified there. I'm really not sure what's going on.

kikaragyozov, 2024-09-18:
I've solved my issue by utilizing the built-in SD copier. Still no clue what went wrong with rpi-clone.

wboerhout, 2024-11-27:
same issue here. rpi-clone works on bookworm an all my raspberries (running the prescribed Debian), but boot fails on the one Raspberry Pi 5.

framps, 2024-11-27:
> all my raspberries

Do you also have a RPi4 and tried to clone a SD card on nmve? 

I unfortunately don't own a RPi5 and cannot try to reproduce your issue. But it sounds to me it's not related to RPi4 but to clone an SD card on a nvme drive. If I'm right the same scenario should will fail if you use a RPi4. Would be great if you can double check with a RPi4.

wboerhout, 2024-11-27:
All works fine on RPi 4. I do not have NVMEs but my targets are the new Samsung USB 3.2 sticks from the PiHutGroet,Wilm(Verstuurd vanaf mijn telefoon, dus wat korter dan gewoonlijk.)(Sent from my phone, so a bit more compact than usual)

framps, 2024-11-27:
Did you test the ssd to nvme move with RPi4? If it fails I'm able to try to reproduce the issue because I have a PRi4 with nvmestorage.

wboerhout, 2024-11-27:
As I said, I have no NVME. I have a Kingston SSD on a USB 3 adapter as source (the Debian system I'm running rpi-clone on, and a Sandisk USB 3.2 USB stick as target. This setup runs fine on RPi4 bookworm (i.e., rpi-clone runs error free and new target will boot the copy)

On Rpi  5, rpi-clone runs error free as well, but the target disk is stuck early in the boot cycle (in a state where I cannot find a log from), but the on screen messages suggest that (the copy of) partition 1 is not bootable for some reason

framps, 2024-11-27:
rpi-clone unfortunately misses logging which makes it - at least for me who has no RPi5 - impossible to recreate and debug your issue :-(

Crossing fingers somebody else owns a RPi5 and can recreate and debug your issue. 

If you're willing to donate to get me to buy a RPi5 - please see [this page](https://www.linux-tips-and-tricks.de/en/donations)

wboerhout, 2024-11-27:
I will think about it. I really appreciate the work you are doing.

framps, 2024-11-28:
I got a donation from @geerlingguy (Thank you very much!!!) and just ordered a RPi5 which will arrive next week. 

No NVMe hat and SSD but I expect your issue is not nvme related and I can reproduce your issue when I clone the SD card to an USB disk attached via USB.

geerlingguy, 2024-11-29:
@framps - Odd! What's your X username? I thought I had DMs open but I guess not!

framps, 2024-11-29:
I just checked my DM settings and it was disabled. It's obviously the default setting. My username is `@linuxframp`.

wboerhout, 2024-11-29:
OK this is the issue more documented. Same process works fine on all Rpi execept Rpi5

```pi@raspi5 ~ $ lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
loop0    7:0    0    92M  1 loop /snap/core/17201
loop1    7:1    0    20K  1 loop /snap/hello-world/29
sda      8:0    0 111,8G  0 disk
├─sda1   8:1    0   512M  0 part /boot/firmware
└─sda2   8:2    0 111,3G  0 part /
sdb      8:16   1  57,3G  0 disk
├─sdb1   8:17   1   512M  0 part
└─sdb2   8:18   1  56,8G  0 part
```
`sda` is my Kingston SATA SSD connected via a SATA/USB adapter to the bottom USB-3 port on the Rpi 5. This is my usual boot setup.
 `sdb` is the new SanDisk Ultra Fit USB 3.2 Flash Drive and the target for rpi-clone and it is in the top USB-3 port.

Now `sudo rpi-clone -f sdb` works without errors, and upon manual inspection, all files seem to be in place. However, booting the new disk fails. See the attached screenshot (literally, with a camera) of what happens. The Rpi 5 keeps trying and the last few lines reappear in what seems to be an endless loop.

[image]

framps, 2024-11-29:
Which OS do you use? Please show the output of `cat /etc-os-release` and `uname -a`.

wboerhout, 2024-11-29:
pi@raspi5 ~ $ cat /etc/os-release
```
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```
pi@raspi5 ~ $ uname -a
```
Linux raspi5 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
```

framps, 2024-11-29:
Ok. Thx. Now I wait formya RPi5 which should arrive mid next week to reproduce and debug this issue :smiley:

wboerhout, 2024-11-29:
since I suspect that it is maybe related to the USB device itself, I include the following info:

pi@raspi5 ~ $ lsusb
```Bus 004 Device 002: ID 0781:5583 SanDisk Corp. Ultra Fit
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 174c:55aa ASMedia Technology Inc. ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
pi@raspi5 ~ $ sudo lsusb -D /dev/bus/usb/004/002
```Device: ID 0781:5583 SanDisk Corp. Ultra Fit
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         9
  idVendor           0x0781 SanDisk Corp.
  idProduct          0x5583 Ultra Fit
  bcdDevice            1.00
  iManufacturer           1  USB
  iProduct                2  SanDisk 3.2Gen1
  iSerial                 3 00023808042724041242
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0079
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              896mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           4
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     98
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
        Command pipe (0x01)
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               3
        MaxStreams              4
        Data-in pipe (0x03)
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
        MaxStreams              4
        Data-out pipe (0x04)
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               3
        MaxStreams              4
        Status pipe (0x02)
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0016
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000002
      HIRD Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat         256 micro seconds
Device Status:     0x0000
  (Bus Powered)
```

framps, 2024-11-29:
> since I suspect that it is maybe related to the USB device

I frankly don't think it's related to the USB device. 

I got a PN in the German Raspberry forum where somebody told me rpi-clone doesn't work for him on a RPi5. Given I don't own a RPi5 .. will own one next week - I wasn't able to help him. He tried to clone form SD card to SD card on an USB adapter. I think there is some basic issue with RPi5 - but I frankly have no idea what's the root cause. I have to recreate the issue on a RPi5 next week.

Did you try another USB device?

wboerhout, 2024-11-30:
I am sorry :-) to report that the issue has disappeared. Frankly, I have no idea why. The clone will now boot successfully.

Even with me, the problem sometimes seems to be located between the chair and the keyboard.

I am happy to assist in further troubleshooting

framps, 2024-11-30:
NP. 

I recently suggested a user in the German Raspberry forum to use rpi-clone to create a clone of his SD card he runs on a RPi5. He also mentioned the cloned SD card didn't boot. Then he executed rpi-clone a second time and then the clone booted. Because I don't have a RPi5 I didn't start to debug the issue. Maybe that's what you also did.

So there seems to be an issue with RPi5 and I will try to reproduce and then debug the it when I have my RPi5.

wboerhout, 2024-11-30:
Possibly yes. It failen a few day ago, but several times. Then this morning, it suddenly went ok.

There may have been a (kernel header?) update in between.

framps, 2024-11-30:
When I came back home today I detected my RPi5 and the required power supply was delivered also today :smiley: So I started immediately to try to reproduce your issue.

I was able to reproduce it twice - but now I'm not able to reproduce it any more to debug the issue :cry: Looks like there is some race condition involved. 

I have no issue on my RPi4 and so I compared the kernel used on RPi4 and RPi5 :

`Linux raspberrypi-bookworm64-lite-rpi5 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux`
and 
`Linux asterix 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux`

Not sure whether there is something different in the kernel which causes the issue.

I noticed the mount update messages - not sure whether they cause the issue.

At least I think there is a temporary workaround available: Execute rpi-clone twice. I know that's inconvenient and hast to be fixed, But a race condition is difficult to fix :cry:

wboerhout, 2024-11-30:
For now, I can live with it. Thanks for your support!

framps, 2024-12-01:
I think I'm one step further. I was able to recreate the issue and detected when rpi-clone finished /boot/firmware wasn't mounted any more. I don't know why this happens intermittently but I added some code [in my fork](https://github.com/framps/rpi-clone/blob/rpi5/rpi-clone) to check whether the partitions are mounted and mount them again if they disappeared.

Everybody who faces the issue should grab my fix and test again. A quick check whether the clone of the boot partition without any reboot can be done with
```
sudo mount <cloneDevice> /mnt
ls -la /mnt
sudo umount /mnt
```
If it's empty the clone will not boot. If somebody still faces the issue with my fix please execute following command
```
sudo ./rpi-clone <options> <cloneDevice> | tee rpi-clone.log
```
This will create a lot of debug output on the screen and in rpi-clone.log. When rpi-clone finished please attach rpi-clone.log in this issue.

framps, 2024-12-01:
Thank you very much again @geerlingguy for your donation which enables me to nail down this issue. 

It's not that I have enough money to buy a RPi5. But it's my principle to to buy any required HW for any OS projects I work on with donations only.

sbhadr, 2024-12-18:
I just so happen to be wanting to move from my microSD to the external SSD. Was there ever a fix? Does the current repo not work? This thread has a lot of words and confusing.  

I am on Rasp pi OS lite (64 bit) on RPi 5. 

@framps @geerlingguy

framps, 2024-12-19:
> Does the current repo not work?

Everything is fine with an RPi4 or older RPi. There is an intermittent issue with RPi5 and therefore I updated rpi-clone in my fork. Please grab [my updated version](https://github.com/framps/rpi-clone/blob/rpi5/rpi-clone) to clone your microSD. 

Please let me know whether the SSD boots or fails to boot. If it fails please show the console output.

framps, 2024-12-23:
@sbhadr Would be great to get an update :wink:

fjpdevries, 2024-12-23:
In verband met de feestdagen ben ik tot en met woensdag 1 januari 2025 afwezig en zal mijn email beperkt kunnen lezen en beantwoorden.

Ik wens u het allerbeste voor 2025 en fijne feestdagen!

Met vriendelijke groet,
FSE Turnstiles b.v.
Jos de Vries

PS: Neem voor urgente storingen contact met ons op door een email te sturen naar ***@***.***

sbhadr, 2024-12-25:
> @sbhadr Would be great to get an update 😉

I just used Balena Etcher, cloned the drive and it picked up the SSD without the microSD attached.

framps, 2024-12-25:
Ok. So you didn't use the updated rpi-clone from my fork to test it works for you - unfortunately. Would be great if you could try the clone again with the updated rpi-clone to verify the fix.

roman-sakh, 2025-01-13:
@framps I am going to test your forked version right away. 
I have only one question, do I need to do mounting for certain partitions? 
Or just cloning should be enough. 

I have RPi5 with HAT pcb and Kingston NVMe SSD installed.
Thank you

framps, 2025-01-13:
> I have only one question, do I need to do mounting for certain partitions?

No. Just use rpi-clone as it's documented. 

If you check my [workaround](https://github.com/framps/rpi-clone/commit/5e68497138e1421db7f0d0c0c6efd4014fbd051c) you'll see I check whether the boot partition is mounted and mount it if it's not mounted.

roman-sakh, 2025-01-13:
this is what i have after cloning 
[image]

I do not like naming in /mnt folder 
and also I do not see MOUNTPOINTS after running lsblk

framps, 2025-01-13:
> I do not like naming in /mnt folder

I see your point. Both mountpoints are created by rpi-clone and should be deleted at the end. I'll add the required rmdir statements in a PR.

> and also I do not see MOUNTPOINTS after running lsblk

Well, rpi-clone assumes that you open another window when you see the inspecting message. Then you will find both partitions mounted. But when you press enter they are unmounted.

Are you sure you use my version? I expect to see the debug output of lsblk which I miss in your screenshot. Please check the md5sum of your version.

```
pi@raspberrypi-bookworm64-lite-beta:~ $ md5sum rpi-clone 
fbeb1ac90bc3ac0d3e2b8716dab722d7  rpi-clone
```

roman-sakh, 2025-01-14:
Sorry for confusion, mountpoints is good. 
rpi-clone is cloning evrything without issues, but RPi is still not  booting from SSD. 
I was doing everything via SSH, cause RPi is in another city =) 
Today my bro will connect HDMI display and will see what it says. 

regarding using your version, yes. I have cloned your repo, switched correct branch. 
Will proceed today or tomorrow and update my findings.

framps, 2025-01-14:
According your screenshot you don't use my version. 

Just use `wget https://raw.githubusercontent.com/framps/rpi-clone/refs/heads/rpi5/rpi-clone` to download the code.

I updated my version yesterday and you now will get also a message it's my version :wink:

roman-sakh, 2025-01-15:
What I have achieved so far. 
I have tried to clone with forked utility from @framps . CLoning is good, RPi5 is boting from SSD to desktop, but after that it is freezing, and screen turns to white background. 
And after that it is just turning to black screen with blinking cursor on the top. 
Seems the problem is in my m.2 HAT which has Switching chip on it. 
Will proceed with findings, but they are not related to cloning process. 

Thank you much for help.

roman-sakh, 2025-01-15:
Latest update. 
@framps  your fork works as charm. 
For some reson I had identical UUID for partiotions in SD-card and SSD. 
After I did everything from scratch it worked from since 1st try. 

What I put in `/boot/firmware/config.txt`
`# Enable the PCIe External connector.
dtparam=pciex1
dtparam=pciex1_gen=2 # may be redundant, but let's be explicit
usb_max_current_enable=1`

Bootloader

`[all]
BOOT_UART=1
BOOT_ORDER=0xf416
#NET_INSTALL_AT_POWER_ON=1
#PCIE_PROBE=1`

framps, 2025-01-15:
> @framps your fork works as charm.

Keep in mind this repository is the master repo for rpi-clone. My for is only required to be able to create PRs to fix issues in rpi-clone :wink: .

evrial, 2025-05-04:
This repo is a 1886 lines wrapper around `dd if=source of=target bs=1M`
Stop using it. Get a card reader, boot linux from other machine and get it done. works smooth.

geerlingguy, 2025-05-04:
@evrial - It's fine to do that if you like. Raspberry Pi OS even includes an SD card copier utility that does a good job as well. Different strokes for different folks :)