geerlingguy/rpi-clone #46: cmdline.txt not update on cloned disk

gtnscebba, 2024-09-24:
I installed version 2.0.22 on a Raspberry PI4b.
The Linux version is SMP PREEMPT Debian 1:6.6.47-1+rpt1 (2024-09-02) aarch64 GNU/Linux.

Cloning is successful, but at the end it only updates fstab and not cmdline.txt.

The result is that the cloned SD is not able to boot correctly.

By manually updating cmdline.txt with the correct PARTUUID, the SD boots correctly. I can see the correct PARTUUID from sudo blkid.

gtnscebba, 2024-09-24:
An update: I've installaed the versione 2.0.23 of rpi-clone and now everything is okay.
rpi-clone updated fstab and cmdline.txt.
Sorry.

gtnscebba, 2024-09-24:
I have installed (by mistake) rpi-clone from https://github.com/billw2/rpi-clone.

framps, 2024-09-24:
Unfortunately it's Bill only who can update his repository and we cannot do anything on this :-(.

fmarzocca, 2024-09-24:
If Bill's repository is not correctly maintaned, it should be closed. This situation generates clouds of confusion. IMHO.

framps, 2024-09-24:
I agree. But I don't think it's possible to close a repository if you're not the owner of the repository.