geerlingguy/rpi-clone #48: Unable to locate boot device

supermavster, 2024-11-18:
`sudo rpi-clone -f sdc`

```bash
 ~ "^/boot(/firmware)?$" &&  == "vfat" { print substr(, 2); exit }
```

Unable to locate boot device

https://github.com/geerlingguy/rpi-clone/blob/deda2cb349801c477fcb3099989398699baf0770/rpi-clone#L946

framps, 2024-11-18:
Would you please show the output of `mount` ?

yoni13, 2024-11-24:
Hello, same error on diet-pi while trying to copy from sdcard to an nvmessd

Command I ran
```
sudo wipefs --all --force /dev/nvme0n1
sudo dd if=/dev/zero of=/dev/nvme0n1 bs=1024 count=1
# Install rpi-clone.
git clone https://github.com/geerlingguy/rpi-clone.git
cd rpi-clone
sudo cp rpi-clone rpi-clone-setup /usr/local/sbin

# Clone to the NVMe drive (usually nvme0n1, but check with `lsblk`).
sudo rpi-clone nvme0n1
```

lsblk
```
root@orangepi3b:~/rpi-clone# lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda           8:0    0 931.5G  0 disk 
└─sda1        8:1    0 931.5G  0 part /srv/dev-disk-by-uuid-8637c3a8-fbd7-410b-a81c-fae73b9ac600
mtdblock0    31:0    0    16M  0 disk 
mmcblk1     179:0    0  29.8G  0 disk 
└─mmcblk1p1 179:1    0  29.7G  0 part /var/folder2ram/var/cache/samba
                                      /var/folder2ram/var/lib/monit
                                      /var/folder2ram/var/lib/rrdcached
                                      /var/folder2ram/var/spool
                                      /var/folder2ram/var/lib/openmediavault/rrd
                                      /var/folder2ram/var/tmp
                                      /
nvme0n1     259:0    0 238.5G  0 disk 
```

And output of `mount`

https://bin.bloerg.net/5Cph2d

framps, 2024-11-25:
rpi-clone expects to have two partitions and /boot/firmware or /boot mounted to partition 1 and / to partition 2.

According your lsblk output diet-pi just uses one partition and that's not supported by rpi-clone.

ccatlett1984, 2025-01-19:
LSBLK output

```
pi@linuxcnc:~/sources/rpi-clone$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk0     179:0    0  58.2G  0 disk
|-mmcblk0p1 179:1    0   508M  0 part /boot/broadcom
`-mmcblk0p2 179:2    0  57.7G  0 part /
zram0       253:0    0     1G  0 disk [SWAP]
nvme0n1     259:0    0 119.2G  0 disk
```

MOUNT output

```

pi@linuxcnc:~/sources/rpi-clone$ mount
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
udev on /dev type devtmpfs (rw,nosuid,relatime,size=4075856k,nr_inodes=254741,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,noexec,relatime,size=824272k,mode=755)
/dev/mmcblk0p2 on / type ext4 (rw,noatime,errors=remount-ro,commit=600)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate,memory_recursiveprot)
pstore on /sys/fs/pstore type pstore (rw,nosuid,nodev,noexec,relatime)
bpf on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,relatime,mode=700)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=30,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=13367)
debugfs on /sys/kernel/debug type debugfs (rw,nosuid,nodev,noexec,relatime)
tracefs on /sys/kernel/tracing type tracefs (rw,nosuid,nodev,noexec,relatime)
mqueue on /dev/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime)
configfs on /sys/kernel/config type configfs (rw,nosuid,nodev,noexec,relatime)
fusectl on /sys/fs/fuse/connections type fusectl (rw,nosuid,nodev,noexec,relatime)
ramfs on /run/credentials/systemd-sysusers.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
ramfs on /run/credentials/systemd-tmpfiles-setup-dev.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
ramfs on /run/credentials/systemd-sysctl.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
tmpfs on /tmp type tmpfs (rw,nosuid,relatime)
/dev/mmcblk0p1 on /boot/broadcom type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,flush,errors=remount-ro)
ramfs on /run/credentials/systemd-tmpfiles-setup.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
binfmt_misc on /proc/sys/fs/binfmt_misc type binfmt_misc (rw,nosuid,nodev,noexec,relatime)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=824272k,nr_inodes=206068,mode=700,uid=1000,gid=1000)
tmpfs on /run/user/108 type tmpfs (rw,nosuid,nodev,relatime,size=824272k,nr_inodes=206068,mode=700,uid=108,gid=117)
```

ccatlett1984, 2025-01-20:
> I'm getting the same error. running standard debian 12
> 
> LSBLK output
> 
> ```
> pi@linuxcnc:~/sources/rpi-clone$ lsblk
> NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
> mmcblk0     179:0    0  58.2G  0 disk
> |-mmcblk0p1 179:1    0   508M  0 part /boot/broadcom
> `-mmcblk0p2 179:2    0  57.7G  0 part /
> zram0       253:0    0     1G  0 disk [SWAP]
> nvme0n1     259:0    0 119.2G  0 disk
> ```
> 
> MOUNT output
> 
> ```
> 
> pi@linuxcnc:~/sources/rpi-clone$ mount
> sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
> proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
> udev on /dev type devtmpfs (rw,nosuid,relatime,size=4075856k,nr_inodes=254741,mode=755)
> devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
> tmpfs on /run type tmpfs (rw,nosuid,nodev,noexec,relatime,size=824272k,mode=755)
> /dev/mmcblk0p2 on / type ext4 (rw,noatime,errors=remount-ro,commit=600)
> securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
> tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
> tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
> cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate,memory_recursiveprot)
> pstore on /sys/fs/pstore type pstore (rw,nosuid,nodev,noexec,relatime)
> bpf on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,relatime,mode=700)
> systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=30,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=13367)
> debugfs on /sys/kernel/debug type debugfs (rw,nosuid,nodev,noexec,relatime)
> tracefs on /sys/kernel/tracing type tracefs (rw,nosuid,nodev,noexec,relatime)
> mqueue on /dev/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime)
> configfs on /sys/kernel/config type configfs (rw,nosuid,nodev,noexec,relatime)
> fusectl on /sys/fs/fuse/connections type fusectl (rw,nosuid,nodev,noexec,relatime)
> ramfs on /run/credentials/systemd-sysusers.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
> ramfs on /run/credentials/systemd-tmpfiles-setup-dev.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
> ramfs on /run/credentials/systemd-sysctl.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
> tmpfs on /tmp type tmpfs (rw,nosuid,relatime)
> /dev/mmcblk0p1 on /boot/broadcom type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,flush,errors=remount-ro)
> ramfs on /run/credentials/systemd-tmpfiles-setup.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
> binfmt_misc on /proc/sys/fs/binfmt_misc type binfmt_misc (rw,nosuid,nodev,noexec,relatime)
> tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=824272k,nr_inodes=206068,mode=700,uid=1000,gid=1000)
> tmpfs on /run/user/108 type tmpfs (rw,nosuid,nodev,relatime,size=824272k,nr_inodes=206068,mode=700,uid=108,gid=117)
> ```

I was able to get around this using DD, just wanted to update for anyone else that runs into this.

ccatlett1984, 2025-01-20:
> @ccatlett1984 <https://github.com/ccatlett1984> You don't use the
> standard name of the boot mountpoint which is /boot/firmware. If you rename
> the boot mountpoint and update /etc/fstab accordingly rpi-clone will work
> for you.

I would suggest a more intuitive option, of adding additional path check in
the code logic to look for this path as well. As many people may be
attempting to use this package, and don't have control over the mount point
for the image that they are using.

mgnthe54, 2025-04-27:
Hello, I'm asking for your advice because I'm having a problem with rpi-clone. I'm also getting the error:
**Unable to locate boot device**
I have a Pi 5 running 64-bit Bookworm with a 128 GB NVMe drive on the PCI port, a 64 GB microSD card, and a 128 GB SSD via USB.
The Pi boots from the NVMe drive. I'd like to make backup copies of this drive to the SSD and then the SD card.
To copy to the SD card, I use this command:
`sudo rpi-clone -l mmcblk0 -v -U`
And to copy to the SSD, I use this command:
`sudo rpi-clone -l sdb1 -v -U`
In both cases, I get the following error:
Unable to locate boot device

`pi@raspberrypi:~ $ sudo blkid /dev/nvme0n1p1: LABEL_FATBOOT="bootfs" LABEL="bootfs" UUID="5DC7-F115" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="4e2eb9a7-01" /dev/nvme0n1p2: LABEL="rootfs" UUID="a36be96c-66be-4487-a7a6-0481bca99d89" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="4e2eb9a7-02" /dev/mmcblk0p1: LABEL_FATBOOT="SD1" LABEL="SD1" UUID="102D-1EC7" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="4e2eb9a7-01" /dev/sr0: BLOCK_SIZE="2048" UUID="2013-11-21-16-45-01-00" LABEL="MegaFon" TYPE="iso9660" PTTYPE="mac" /dev/sdb2: LABEL="SSD2" UUID="a8166df1-14f7-4e91-8818-c9b600e3d111" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="e40ddcee-02" /dev/sdb1: LABEL_FATBOOT="SSD1" LABEL="SSD1" UUID="703F-9483" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="e40ddcee-01" /dev/mmcblk0p2: LABEL="SD2" UUID="d18b23e9-5237-4d2e-a116-599d5602a720" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="4e2eb9a7-02" `
Do you know how to solve this problem? Did I make a mistake?

`pi@raspberrypi:~ $ sudo mount
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
udev on /dev type devtmpfs (rw,nosuid,relatime,size=8260416k,nr_inodes=516276,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,noexec,relatime,size=1660480k,mode=755)
/dev/nvme0n1p2 on / type ext4 (rw,noatime)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate,memory_recursiveprot)
pstore on /sys/fs/pstore type pstore (rw,nosuid,nodev,noexec,relatime)
bpf on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,relatime,mode=700)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=29,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=1247)
debugfs on /sys/kernel/debug type debugfs (rw,nosuid,nodev,noexec,relatime)
mqueue on /dev/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime)
tracefs on /sys/kernel/tracing type tracefs (rw,nosuid,nodev,noexec,relatime)
fusectl on /sys/fs/fuse/connections type fusectl (rw,nosuid,nodev,noexec,relatime)
configfs on /sys/kernel/config type configfs (rw,nosuid,nodev,noexec,relatime)
ramfs on /run/credentials/systemd-sysctl.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
ramfs on /run/credentials/systemd-sysusers.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
ramfs on /run/credentials/systemd-tmpfiles-setup-dev.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
systemd-1 on /mnt/Medion26 type autofs (rw,relatime,fd=41,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=2549)
systemd-1 on /mnt/share type autofs (rw,relatime,fd=42,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=2553)
**/dev/nvme0n1p1 on /boot/firmware type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/nvme0n1p2 on /var/hdd.log type ext4 (rw,noatime)**
binfmt_misc on /proc/sys/fs/binfmt_misc type binfmt_misc (rw,nosuid,nodev,noexec,relatime)
log2ram on /var/log type tmpfs (rw,nosuid,nodev,noexec,noatime,size=524288k,mode=755)
ramfs on /run/credentials/systemd-tmpfiles-setup.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=1660464k,nr_inodes=415116,mode=700,uid=1000,gid=1000)
gvfsd-fuse on /run/user/1000/gvfs type fuse.gvfsd-fuse (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
portal on /run/user/1000/doc type fuse.portal (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
/dev/sda2 on /media/pi/SSD2 type ext4 (rw,nosuid,nodev,relatime,errors=remount-ro,uhelper=udisks2)
/dev/sda1 on /media/pi/SSD1 type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
/dev/sr0 on /media/pi/MegaFon type iso9660 (ro,nosuid,nodev,relatime,nojoliet,check=s,map=n,blocksize=2048,uid=1000,gid=1000,dmode=500,fmode=400,iocharset=utf8,uhelper=udisks2)`
Thank you.

jeshtroy, 2025-08-16:
Hey @framps, firstly, thanks for helping with maintaining this tool! I'm getting this same error. What should I doing exactly in my situation? `mount` command doesn't have boot anywhere in it

`
$ lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
sda           8:0    1 29.1G  0 disk
├─sda1        8:1    1  512M  0 part
└─sda2        8:2    1 28.6G  0 part
mmcblk0     179:0    0 29.5G  0 disk
├─mmcblk0p1 179:1    0  512M  0 part
└─mmcblk0p2 179:2    0   29G  0 part /
`

`
$ cat /etc/fstab
proc            /proc           proc    defaults          0       0
PARTUUID=c795axxx-01  /boot/firmware  vfat    defaults          0       2
PARTUUID=c795axxx-02  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
`