geerlingguy/rpi-clone #49: Raspi5 - Cloning from PCI NVMe to SD card slot. Raspi do not boot (B1) from SD card.

Manu-65, 2024-12-01:
Hello,
i have a raspi5 with nvme pci hat.
At the current the raspi5 is booting from nvme (no SD card inserted in raspi slot)

I would like to clone the nvme to SD card, then i inserted a SD card.
I launched rpi-clone but after that raspi5 is not booting from the SD card (boot order is B1, Boot from SD Card before trying NVMe...)

I am not familiar with rpi-clone command (i'm beginner...)

I tried the next commands:
~# rpi-clone -f2 mmcblk0
and
~# rpi-clone -f2 -l mmcblk0

i'm not sure what i'm doing...
which command options should i use?

Any idea?
Thanks
Manuel

~# lsblk

NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk0     179:0    0  59.2G  0 disk 
├─mmcblk0p1 179:1    0   512M  0 part 
└─mmcblk0p2 179:2    0  58.7G  0 part 
nvme0n1     259:0    0 465.8G  0 disk 
├─nvme0n1p1 259:1    0   512M  0 part /boot/firmware
├─nvme0n1p2 259:2    0  58.7G  0 part /
├─nvme0n1p3 259:3    0  58.7G  0 part /mnt/pve/backup
└─nvme0n1p4 259:4    0 347.8G  0 part /mnt/pve/datastore

~# cat /etc/os-release

PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

framps, 2024-12-01:
Please see https://github.com/geerlingguy/rpi-clone/issues/44 and try the workaround.

Manu-65, 2024-12-01:
Thanks, i saw #44, today i repeated the process (commands) four times, there is no change/improvement. 

~# uname -a

Linux 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux

framps, 2024-12-01:
I'm just working on this nasty issue and I was able to fix the issue when I ran rpi-clone a second time. Unfortunately I can't reproduce the issue reliable which is quite disappointing :disappointed: 

Please try again but don't use `-f` or `-f2`. Just call `sudo rpi-clone <device>`

Manu-65, 2024-12-01:
Without -f2, it is not able to clone.                          

~# rpi-clone -v mmcblk0

Booted disk: nvme0n1 500.1GB               Destination disk: mmcblk0 63.6GB
---------------------------------------------------------------------------
Part                                Size     FS        Label           Part     Size         FS        Label  
1 /boot/firmware       512.0M    fat32    --                1         512.0M  fat32    --     
2 root                          58.7G    ext4     rootfs           2         58.7G     ext4     --     
3 /mnt/pve/backup       58.7G   ext4     --                                           
4 /mnt/pve/datastore   347.8G  ext4     --                                           
---------------------------------------------------------------------------
** FATAL **            : Initialize needed - partition number mismatch: 4 -> 2
                       : But destination is too small to clone 4 partitions.
                       : Minimum destination size required is 117.9G.
                       : Possible options:
                       :     Use -f2 to force a two partition initialize clone.
                       :     Use -m to limit partitions to clone.
                       :     Manually create custom partitions that can work.
                       :     Larger destination disk..
-----------------------:

framps, 2024-12-01:
Good catch :+1:  Looks like I should also try a clone to a smaller device to reproduce the issue. I'm unfortunately no longer able to reproduce this issue. It's a really ugly issue ...

Manu-65, 2024-12-01:
[image]

=> rsync // /mnt/clone with-root-excludes...    is it good?

framps, 2024-12-01:
> i read that in previous versions some files (cmdline.txt, config.txt) were saved in /boot folder and at the current version these files are saved in /boot/firmware.
> 
> could be it the root of this issue?

No. 

> i have cloned it but it seems not reliable.

Yes unfortunately :cry: It seems to be an issue with the target device speed and/or other Kernel used and the fact RPi5 is faster than a RPi4.

framps, 2024-12-01:
@Manu-65 Do you remember what you did in order to get it succeed? Just tried multiple times or what else?

framps, 2024-12-01:
@Manu-65 That's OK with root-exclude.

> should i edit some file? fstab, etc...

No. But if you test my fix I mentioned in #44 that would be great.

Manu-65, 2024-12-01:
@framps Thanks for your feedback

mgnthe54, 2025-04-26:
Hello, I'm asking for your advice because I'm having a problem with rpi-clone. I'm also getting the error:
**Unable to locate boot device**
I have a Pi 5 running 64-bit Bookworm with a 128 GB NVMe drive on the PCI port, a 64 GB microSD card, and a 128 GB SSD via USB.
The Pi boots from the NVMe drive. I'd like to make backup copies of this drive to the SSD and then the SD card.
To copy to the SD card, I use this command:
`sudo rpi-clone -l mmcblk0 -v -U`
And to copy to the SSD, I use this command:
`sudo rpi-clone -l sdb1 -v -U`
In both cases, I get the following error:
Unable to locate boot device

`pi@raspberrypi:~ $ sudo blkid
/dev/nvme0n1p1: LABEL_FATBOOT="bootfs" LABEL="bootfs" UUID="5DC7-F115" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="4e2eb9a7-01"
/dev/nvme0n1p2: LABEL="rootfs" UUID="a36be96c-66be-4487-a7a6-0481bca99d89" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="4e2eb9a7-02"
/dev/mmcblk0p1: LABEL_FATBOOT="SD1" LABEL="SD1" UUID="102D-1EC7" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="4e2eb9a7-01"
/dev/sr0: BLOCK_SIZE="2048" UUID="2013-11-21-16-45-01-00" LABEL="MegaFon" TYPE="iso9660" PTTYPE="mac"
/dev/sdb2: LABEL="SSD2" UUID="a8166df1-14f7-4e91-8818-c9b600e3d111" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="e40ddcee-02"
/dev/sdb1: LABEL_FATBOOT="SSD1" LABEL="SSD1" UUID="703F-9483" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="e40ddcee-01"
/dev/mmcblk0p2: LABEL="SD2" UUID="d18b23e9-5237-4d2e-a116-599d5602a720" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="4e2eb9a7-02"
`
Do you know how to solve this problem? Did I make a mistake?
Thank you.