geerlingguy/rpi-clone #55: Cannot be used on encrypted pi(?)

Tryfos, 2025-01-29:
I'm trying to use rpi-clone on an encrypted pi4 after having decrypted the partition with a network passphrase with dropbear and booted succesfully. I used https://github.com/gitbls/sdm in order to encrypt both partitions of a pios .img and burn to a boot SSD. Now I'm trying to clone SSD to SD. My root partition is named cryptroot and is mounted on /dev/mapper/cryptroot. As soon as I enter sudo rpi-clone to simply get the man page I get "Error: Could not stat device /dev/mapper/cryptroo - No such file or directory."  Funny thing it says "cryptroo", not even cryptroot. I searched the file system and I see that /dev/mapper/cryptroot is a hard link "cryptroot -> ../dm-0", so perhaps that's why it cannot be located by rpi-clone.
a) I'm I right in assuming that encrypted partitions are not supported by rpi-clone? I thought that as soon as the filesystem is decrypted it has no major differencies to an unencrypted one. b) I think that the "man page" (rpi-clone without arguments) should work regardless. c) Why does the error say "cryptroo" instead of cryptroot?
Thanks in advance.

Tryfos, 2025-02-07:
Anyone, any ideas?

CTLQ-dev, 2025-04-02:
I know its a little late but I have been trying to achieve the same thing with extreme difficulty

closed 2026-04-25 by framps