geerlingguy/rpi-clone #56: How to restore from backup recovery in ssd to sd card

zskangus, 2025-01-30:
I have a raspberry pi 5 with a ssd card, and a 1t nvme ssd via hat extension.

 The ssd card is for operating system and ssd is for storing files, so far I have realized to backup ssd card to ssd.

 How do I restore a backup from ssd to sdcard? This can be done without any external tools and without plugging or unplugging the ssd card

framps, 2025-01-30:
When you have cloned the SD card to your SSD you will be able to boot from your SSD the clone. Then use rpi-clone to clone the clone to any SD card.

zskangus, 2025-02-01:
> When you have cloned the SD card to your SSD you will be able to boot from your SSD the clone. Then use rpi-clone to clone the clone to any SD card.

I wrote a switching script directly for ssd and ssd startup switching, and can easily use the clone now

framps, 2026-04-25:
Great you found a solution. May you want to share your solution?

closed 2026-04-25 by framps