geerlingguy/rpi-clone #64: Is -L lbl option with a trailing # no longer supported or broken?

mys10v8, 2025-07-17:
The -L option with a trailing '#' in the label used to substitute a partition number for the pound sign.  Doesn't do that anymore.  Below are samples from an old version of rpi-clone and the current.

old rpi-clone script working example:
```
$ sudo rpi-clone -f -L `date +"MN5%y%m%d_#"` sdb

Booted disk: sda 240.1GB                   Destination disk: sdb 63.9GB
---------------------------------------------------------------------------
Part      Size    FS     Label           Part   Size    FS     Label   
1 /boot   256.0M  fat32  --              1      512.0M  fat32  --      
2 root     14.6G  ext4   mn240518_2      2        2.1G  ext4   rootfs  
3 /big    208.7G  ext4   mn240518_3                                    
---------------------------------------------------------------------------
== Initialize: IMAGE partition table - forced by option ==
1 /boot               (51.5M used)   : MKFS  SYNC to sdb1
2 root                (6.9G used)    : MKFS  SYNC to sdb2
3 /big                (35.0G used)   : RESIZE  MKFS  SYNC to sdb3
---------------------------------------------------------------------------
-L MN5250717_#         : volume label for all ext partition types.
Run setup script       : no.
Verbose mode           : no.
-----------------------:
** WARNING **          : All destination disk sdb data will be overwritten!
-----------------------:

Initialize and clone to the destination disk sdb?  (yes/no): yes

Initializing
  Imaging past partition 1 start.
  => dd if=/dev/sda of=/dev/sdb bs=1M count=8 ...
  Resizing destination disk last partition ...
    Resize success.
  Changing destination Disk ID ...
  => mkfs -t vfat -F 32  /dev/sdb1 ...
  => mkfs -t ext4 -L MN5250717_2 /dev/sdb2 ...
  => mkfs -t ext4 -L MN5250717_3 /dev/sdb3 ...

Syncing file systems (can take a long time)
Syncing mounted partitions:
  Mounting /dev/sdb2 on /mnt/clone
  => rsync // /mnt/clone with-root-excludes ...
  Mounting /dev/sdb1 on /mnt/clone/boot
  => rsync /boot/ /mnt/clone/boot  ...
  Mounting /dev/sdb3 on /mnt/clone/big
  => rsync /big/ /mnt/clone/big  ...

Editing /mnt/clone/boot/cmdline.txt PARTUUID to use 27613bbd
Editing /mnt/clone/etc/fstab PARTUUID to use 27613bbd
===============================
Done with clone to /dev/sdb
   Start - 07:37:19    End - 08:17:51    Elapsed Time - 40:32

Cloned partitions are mounted on /mnt/clone for inspection or customizing. 

Hit Enter when ready to unmount the /dev/sdb partitions

$ lsblk -o NAME,LABEL,FSTYPE,FSSIZE,FSUSE%,MOUNTPOINT,PARTUUID
NAME        LABEL       FSTYPE FSSIZE FSUSE% MOUNTPOINT PARTUUID
sda                                                     
├─sda1                  vfat     252M    20% /boot      c17e521a-01
├─sda2      mn240518_2  ext4    14.3G    48% /          c17e521a-02
└─sda3      mn240518_3  ext4   204.3G    26% /big       c17e521a-03
sdb                                                     
├─sdb1                  vfat                            27613bbd-01
├─sdb2      MN5250717_2 ext4                            27613bbd-02
└─sdb3      MN5250717_3 ext4                            27613bbd-03
mmcblk0                                                 
├─mmcblk0p1             vfat                            4d89c080-01
├─mmcblk0p2 tx240922_2  ext4                            4d89c080-02
└─mmcblk0p3 tx240922_3  ext4                            4d89c080-03
```
__________________________________________________________________________________________
current rpi-clone broken example:
```
$ sudo rpi-clone -f -L `date +"MN%y%m%d_#"` sda

Booted disk: mmcblk0 63.9GB                Destination disk: sda 63.9GB
---------------------------------------------------------------------------
Part               Size    FS     Label           Part   Size    FS     Label   
1 /boot/firmware   512.0M  fat32  bootfs          1      512.0M  fat32  bootfs  
2 root              14.6G  ext4   rootfs          2       14.6G  ext4   rootfs  
3 /big              44.3G  ext4   big             3       44.3G  ext4   bigfs   
---------------------------------------------------------------------------
== Initialize: IMAGE partition table - forced by option ==
1 /boot/firmware      (99.0M used)   : MKFS  SYNC to sda1
2 root                (4.7G used)    : MKFS  SYNC to sda2
3 /big                (949.5M used)  : RESIZE  MKFS  SYNC to sda3
---------------------------------------------------------------------------
-L MN250717_#          : volume label for all ext partition types.
Run setup script       : no.
Verbose mode           : no.
-----------------------:
** WARNING **          : All destination disk sda data will be overwritten!
-----------------------:

Initialize and clone to the destination disk sda?  (yes/no): yes

Initializing
  Imaging past partition 1 start.
  => dd if=/dev/mmcblk0 of=/dev/sda bs=1M count=8 ...
  Resizing destination disk last partition ...
    Resize success.
  Changing destination Disk ID ...
  => mkfs -t vfat -F 32 -n bootfs /dev/sda1 ...
  => mkfs -t ext4 -L **MN250717_** /dev/sda2 ...
  => mkfs -t ext4 -L **MN250717_** /dev/sda3 ...

Syncing file systems (can take a long time)
Syncing mounted partitions:
     e2label /dev/sda2 rootfs
  Mounting /dev/sda2 on /mnt/clone
  => rsync // /mnt/clone with-root-excludes ...
     fatlabel /dev/sda1 bootfs
fatlabel: warning - lowercase labels might not work properly on some systems
  Mounting /dev/sda1 on /mnt/clone/boot/firmware
  => rsync /boot/firmware/ /mnt/clone/boot/firmware  ...  Remounting /boot/firmware/

     e2label /dev/sda3 big
  Mounting /dev/sda3 on /mnt/clone/big
  => rsync /big/ /mnt/clone/big  ...

Editing /mnt/clone/boot/firmware/cmdline.txt PARTUUID to use 1a24021b
Editing /mnt/clone/etc/fstab PARTUUID to use 1a24021b
===============================
Done with clone to /dev/sda
   Start - 17:00:59    End - 17:08:29    Elapsed Time - 7:30

Cloned partitions are mounted on /mnt/clone for inspection or customizing.

Hit Enter when ready to unmount the /dev/sda partitions 

$ lsblk -o NAME,LABEL,FSTYPE,FSSIZE,FSUSE%,MOUNTPOINT,PARTUUID
NAME        LABEL  FSTYPE FSSIZE FSUSE% MOUNTPOINT     PARTUUID
sda                                                    
├─sda1      bootfs vfat                                1a24021b-01
├─sda2      rootfs ext4                                1a24021b-02
└─sda3      big    ext4                                1a24021b-03
mmcblk0                                                
├─mmcblk0p1 bootfs vfat     510M    19% /boot/firmware d28ec40f-01
├─mmcblk0p2 rootfs ext4    14.4G    33% /              d28ec40f-02
└─mmcblk0p3 big    ext4    43.3G     2% /big           d28ec40f-03
```

mys10v8, 2025-07-26:
Adding partition number in dst_part_label() is just part 1 of the problem.  

Part 2 of the problem is that dst_part_label() is called twice.  Once for mkfs and again for fatlabel/e2label.  The mkfs commands create the labels I'm expecting but then the fatlabel/e2label change them to something else because by that time ext_label has changed value (to an empty string) and a different label is returned by dst_part_label().  This can be seen in the output below.

Here's my changes in dst_part_label() with pnum addition and debug messages:

```
dst_part_label()                
        {
        pnum=$1
        fs_type=$2
        label=""
                
zzz=${src_label[$pnum]}         
for xxx in ext_label fs_type root_part_num pnum zzz "3" label; do printf "entry %-10s=|%s|\n" $xxx ${!xxx}; done;

        if [ "$ext_label" != "" ] && [[ "$fs_type" == *"ext"* ]]
        then
                rep="${ext_label: -1}"
                if [ "$rep" == "#" ]
                then
                        label="${ext_label:: -1}$pnum"
                elif ((pnum == root_part_num))
                then    
                        label="$ext_label"
                fi
        fi                      
        
        if [ -z "$label" -a -n "${src_label[$pnum]}" ]
        then
                label="${src_label[$pnum]}"
        fi      
        printf -v "${3}" "%s" "$label"
zzz=${src_label[$pnum]} 
for xxx in ext_label fs_type root_part_num pnum zzz "3" label; do printf "exit  %-10s=|%s|\n" $xxx ${!xxx}; done;
        }       
```

And here is the output:

```
$ sudo rpi-clone.dbg -f -L `date +"MN%y%m%d_#"` sda

Booted disk: mmcblk0 63.9GB                Destination disk: sda 63.9GB
---------------------------------------------------------------------------
Part               Size    FS     Label           Part   Size    FS     Label   
1 /boot/firmware   512.0M  fat32  bootfs          1      512.0M  fat32  bootfs  
2 root              14.6G  ext4   rootfs          2       14.6G  ext4   rootfs  
3 /big              44.3G  ext4   big             3       44.3G  ext4   big     
---------------------------------------------------------------------------
== Initialize: IMAGE partition table - forced by option ==
1 /boot/firmware      (99.0M used)   : MKFS  SYNC to sda1
2 root                (4.7G used)    : MKFS  SYNC to sda2
3 /big                (949.5M used)  : RESIZE  MKFS  SYNC to sda3
---------------------------------------------------------------------------
-L MN250725_#          : volume label for all ext partition types.
Run setup script       : no.
Verbose mode           : no.
-----------------------:
** WARNING **          : All destination disk sda data will be overwritten!
-----------------------:

Initialize and clone to the destination disk sda?  (yes/no): yes

Initializing
  Imaging past partition 1 start.
  => dd if=/dev/mmcblk0 of=/dev/sda bs=1M count=8 ...
  Resizing destination disk last partition ...
    Resize success.
  Changing destination Disk ID ...
[... 91 lines cut ...]
exit  fs_type   =|ext4|
exit  root_part_num=|2|
exit  pnum      =|3|
exit  zzz       =|big|
exit  3         =|label|
exit  label     =|big|
     e2label /dev/sda3 big
  Mounting /dev/sda3 on /mnt/clone/big
  => rsync /big/ /mnt/clone/big  ...

Editing /mnt/clone/boot/firmware/cmdline.txt PARTUUID to use 6d915c90
Editing /mnt/clone/etc/fstab PARTUUID to use 6d915c90
===============================
Done with clone to /dev/sda
   Start - 20:06:31    End - 20:14:01    Elapsed Time - 7:30

Cloned partitions are mounted on /mnt/clone for inspection or customizing.

Hit Enter when ready to unmount the /dev/sda partitions ...
  unmounting /mnt/clone/big
  unmounting /mnt/clone/boot/firmware
  unmounting /mnt/clone
===============================

```

matthijskooijman, 2025-07-26:
> [Here in your PR ](https://github.com/geerlingguy/rpi-clone/commit/16f99d991cb95af459f7fa9139bbc701cc239679#diff-e73eaae36cd02c1c899d7257d01e6ad2de891b767399796c10ef7458d271784cL560)in line 561 the partition number is appended in the old code but not in the new line 559.

Ah, good catch. I must have overlooked that `$pnum` in there when refactoring the code. Fixing PR looks good to me, thanks!

> Part 2 of the problem is that dst_part_label() is called twice. Once for mkfs and again for fatlabel/e2label. The mkfs commands create the labels I'm expecting but when called for fatlabel/e2label it returns something else because by that time ext_label has changed value (to an empty string) and a different label is returned by dst_part_label(). This can be seen in the output below.

Indeed, your debug output is helpful, thanks.

It seems that `ext_label` is cleared here:

https://github.com/geerlingguy/rpi-clone/blob/8edb4f8de021f88757b78d2dd785d45b1f109276/rpi-clone#L1634

This happens only when initializing, and after full filesystems are created or dd'd but before syncing filesystems (and copying filesystem labels).

TBH, I am not sure why clearing `ext_label` is needed here (maybe it is just legacy). It was introduced here: https://github.com/geerlingguy/rpi-clone/commit/fe7a646261d2add97928918bafd16a1cac70eb7c Maybe to prevent setting the label twice? I guess we could just remove that line? Or instead of using `$ext_label` in `dst_part_label()`, process it earlier and update the `src_label` array for partitions where it applies (maybe combined with making `change_label()` smarter to only change the label when it needs changing)? But just removing the line is probably easiest :-)

framps, 2025-07-26:
You're our label guru 😉 I'm frankly lost. 

> But just removing the line is probably easiest :-)

I executed rpi-clone, kept line 1634 and got following result:
```
pi@raspberrypi-bookworm64-lite:~ $ lsblk -o NAME,LABEL,FSTYPE,FSSIZE,FSUSE%,MOUNTPOINT,PARTUUID
NAME   LABEL      FSTYPE FSSIZE FSUSE% MOUNTPOINT     PARTUUID
sda                                                   
sdb                                                   
|-sdb1 bootfs     vfat     510M    11% /boot/firmware b01c1e4b-01
|-sdb2            ext4     4.7G    51% /              b01c1e4b-02
`-sdb3            ext4                                b01c1e4b-03
sdc                                                   
|-sdc1 bootfs     vfat                                801b8600-01
|-sdc2 --         ext4                                801b8600-02
`-sdc3 MN250726_3 ext4                                801b8600-03
sdd                         
```
Then I commented out line 1634 and got
```
pi@raspberrypi-bookworm64-lite:~ $ lsblk -o NAME,LABEL,FSTYPE,FSSIZE,FSUSE%,MOUNTPOINT,PARTUUID
NAME   LABEL      FSTYPE FSSIZE FSUSE% MOUNTPOINT     PARTUUID
sda                                                   
sdb                                                   
|-sdb1 bootfs     vfat     510M    11% /boot/firmware b01c1e4b-01
|-sdb2            ext4     4.7G    51% /              b01c1e4b-02
`-sdb3            ext4                                b01c1e4b-03
sdc                                                   
|-sdc1 bootfs     vfat                                095ba699-01
|-sdc2 MN250726_2 ext4                                095ba699-02
`-sdc3 MN250726_3 ext4                                095ba699-03
sdd       
```

So from a testing point of view your proposed easy fix works 👍 I suggest @mys10v8 also tests your fix and if it works for him also you (@matthijskooijman) create this tiny PR to fix this issue. Agreed?

mys10v8, 2025-07-27:
Tested with 1634 commented.  Works for me.  Left debug messages in:

```
$ sudo rpi-clone.dbg -f -L `date +"MN%y%m%d_#"` sda

Booted disk: mmcblk0 63.9GB                Destination disk: sda 63.9GB
---------------------------------------------------------------------------
Part               Size    FS     Label           Part   Size    FS     Label   
1 /boot/firmware   512.0M  fat32  bootfs          1      512.0M  fat32  bootfs  
2 root              14.6G  ext4   rootfs          2       14.6G  ext4   rootfs  
3 /big              44.3G  ext4   big             3       44.3G  ext4   big     
---------------------------------------------------------------------------
== Initialize: IMAGE partition table - forced by option ==
1 /boot/firmware      (99.0M used)   : MKFS  SYNC to sda1
2 root                (4.7G used)    : MKFS  SYNC to sda2
3 /big                (949.5M used)  : RESIZE  MKFS  SYNC to sda3
---------------------------------------------------------------------------
-L MN250727_#          : volume label for all ext partition types.
Run setup script       : no.
Verbose mode           : no.
-----------------------:
** WARNING **          : All destination disk sda data will be overwritten!
-----------------------:

Initialize and clone to the destination disk sda?  (yes/no): yes

Initializing
  Imaging past partition 1 start.
  => dd if=/dev/mmcblk0 of=/dev/sda bs=1M count=8 ...
  Resizing destination disk last partition ...
    Resize success.
  Changing destination Disk ID ...
entry ext_label =|MN250727_#|
entry fs_type   =|fat32|
entry root_part_num=|2|
entry pnum      =|1|
entry zzz       =|bootfs|
entry 3         =|label|
entry label     =||
exit  ext_label =|MN250727_#|
exit  fs_type   =|fat32|
exit  root_part_num=|2|
exit  pnum      =|1|
exit  zzz       =|bootfs|
exit  3         =|label|
exit  label     =|bootfs|
  => mkfs -t vfat -F 32 -n bootfs /dev/sda1 ...
entry ext_label =|MN250727_#|
entry fs_type   =|ext4|
entry root_part_num=|2|
entry pnum      =|2|
entry zzz       =|rootfs|
entry 3         =|label|
entry label     =||
exit  ext_label =|MN250727_#|
exit  fs_type   =|ext4|
exit  root_part_num=|2|
exit  pnum      =|2|
exit  zzz       =|rootfs|
exit  3         =|label|
exit  label     =|MN250727_2|
  => mkfs -t ext4 -L MN250727_2 /dev/sda2 ...
entry ext_label =|MN250727_#|
entry fs_type   =|ext4|
entry root_part_num=|2|
entry pnum      =|3|
entry zzz       =|big|
entry 3         =|label|
entry label     =||
exit  ext_label =|MN250727_#|
exit  fs_type   =|ext4|
exit  root_part_num=|2|
exit  pnum      =|3|
exit  zzz       =|big|
exit  3         =|label|
exit  label     =|MN250727_3|
  => mkfs -t ext4 -L MN250727_3 /dev/sda3 ...

Syncing file systems (can take a long time)
Syncing mounted partitions:
entry ext_label =|MN250727_#|
entry fs_type   =|ext4|
entry root_part_num=|2|
entry pnum      =|2|
entry zzz       =|rootfs|
entry 3         =|label|
entry label     =||
exit  ext_label =|MN250727_#|
exit  fs_type   =|ext4|
exit  root_part_num=|2|
exit  pnum      =|2|
exit  zzz       =|rootfs|
exit  3         =|label|
exit  label     =|MN250727_2|
     e2label /dev/sda2 MN250727_2
  Mounting /dev/sda2 on /mnt/clone
  => rsync // /mnt/clone with-root-excludes ...
entry ext_label =|MN250727_#|
entry fs_type   =|fat32|
entry root_part_num=|2|
entry pnum      =|1|
entry zzz       =|bootfs|
entry 3         =|label|
entry label     =||
exit  ext_label =|MN250727_#|
exit  fs_type   =|fat32|
exit  root_part_num=|2|
exit  pnum      =|1|
exit  zzz       =|bootfs|
exit  3         =|label|
exit  label     =|bootfs|
     fatlabel /dev/sda1 bootfs
fatlabel: warning - lowercase labels might not work properly on some systems
  Mounting /dev/sda1 on /mnt/clone/boot/firmware
  => rsync /boot/firmware/ /mnt/clone/boot/firmware  ...
entry ext_label =|MN250727_#|
entry fs_type   =|ext4|
entry root_part_num=|2|
entry pnum      =|3|
entry zzz       =|big|
entry 3         =|label|
entry label     =||
exit  ext_label =|MN250727_#|
exit  fs_type   =|ext4|
exit  root_part_num=|2|
exit  pnum      =|3|
exit  zzz       =|big|
exit  3         =|label|
exit  label     =|MN250727_3|
     e2label /dev/sda3 MN250727_3
  Mounting /dev/sda3 on /mnt/clone/big
  => rsync /big/ /mnt/clone/big  ...

Editing /mnt/clone/boot/firmware/cmdline.txt PARTUUID to use 1ea74641
Editing /mnt/clone/etc/fstab PARTUUID to use 1ea74641
===============================
Done with clone to /dev/sda
   Start - 07:18:54    End - 07:26:21    Elapsed Time - 7:27

Cloned partitions are mounted on /mnt/clone for inspection or customizing.

Hit Enter when ready to unmount the /dev/sda partitions ...
  unmounting /mnt/clone/big
  unmounting /mnt/clone/boot/firmware
  unmounting /mnt/clone
===============================
```

```
NAME        LABEL      FSTYPE FSSIZE FSUSE% MOUNTPOINT     PARTUUID
sda                                                        
├─sda1      bootfs     vfat                                1ea74641-01
├─sda2      MN250727_2 ext4                                1ea74641-02
└─sda3      MN250727_3 ext4                                1ea74641-03
mmcblk0                                                    
├─mmcblk0p1 bootfs     vfat     510M    19% /boot/firmware d28ec40f-01
├─mmcblk0p2 rootfs     ext4    14.4G    33% /              d28ec40f-02
└─mmcblk0p3 big        ext4    43.3G     2% /big           d28ec40f-03
```

commit 9d07a8c5 (2025-08-22): Merge pull request #68 from framps/m_64

Fixed #64
commit efb6d3ae (2025-08-22): Fixed #64
closed 2025-08-22 by geerlingguy