geerlingguy/rpi-clone #67: Clone has no /boot/firmware directory

Lightwel, 2025-08-20:
Thanks to @geerlingguy for this.

I want to use rpi-clone to clone first two partitions of my RPi5 NVME to SSD to act as a bootable recovery disk in case I stuff the rootfs partition installing BVM (thanks for the video on that as well BTW).

$ sudo rpi-clone -f2 sda seemed to work but on checking /mnt/clone/boot I get cmdline.txt in /boot instead of in boot/firmware as expected.

Additionally, in mnt/clone there is another /mnt/clone which is however empty.

Am I doing something wrong?

Have not tried booting from the new sda pending an answer to this and also the following question which I think ought to be clear to me but isn't:

in order to get a bootable recovery disk do I have to change the partuuid's on the sda clone, and in order to boot the sda clone and then restore to the original NVME do I have to change the partuuid's again after restoration? (The original NVME includes a data partition as well as the two RPi partitions. I can backup the data partition separately, if that is a simpler solution.)

Thanks for any help

Lightwell

framps, 2025-08-21:
It sounds you use an outdated version of rpi-clone. When they changed the location of boot with Bookworm rpi-clone was updated.

> do I have to change the partuuid's again after restoration? 

No

Lightwel, 2025-08-21:
Tried the script again and this time it produced the /boot/firmware directory containing cmdline.txt.  

Don'tknow what went wrong the first time but apologies for the distraction. 

Best
L