geerlingguy/rpi-clone #70: Sometimes /boot/firmware is empty after cloning Ubuntu on RPi5

moujikov, 2025-09-27:
Hello, 
I already ran into this issue couple of weeks ago and once again today. It did not reproduce after subsequent clone though.
It happens as follows:

### Step 1: Cloning Ubuntu from SDA to NVME on RPi5

```
andrey@raspberry-2:/home/andrey$ sudo rpi-clone/rpi-clone nvme0n1 -f -L writable
Error: /dev/nvme0n1: unrecognised disk label

Booted disk: sda 500.1GB                   Destination disk: nvme0n1 512.1GB
---------------------------------------------------------------------------
Part               Size    FS     Label           Part   Size  FS  Label
1 /boot/firmware   512.0M  fat32  system-boot
2 root             465.3G  ext4   writable
---------------------------------------------------------------------------
== Initialize: IMAGE partition table - forced by option ==
1 /boot/firmware      (182.5M used)  : MKFS  SYNC to nvme0n1p1
2 root                (13.6G used)   : RESIZE  MKFS  SYNC to nvme0n1p2
---------------------------------------------------------------------------
-e clone fstab edit    : edit sda device entries to nvme0n1p.
-L writable            : volume label for root partition only.
Run setup script       : no.
Verbose mode           : no.
-----------------------:
** WARNING **          : All destination disk nvme0n1 data will be overwritten!
-----------------------:

Initialize and clone to the destination disk nvme0n1?  (yes/no): y

Initializing
  Imaging past partition 1 start.
  => dd if=/dev/sda of=/dev/nvme0n1 bs=1M count=5 ...
  Resizing destination disk last partition ...
    Resize success.
  Changing destination Disk ID ...
  => mkfs -t vfat -F 32 -n system-boot /dev/nvme0n1p1 ...
  => mkfs -t ext4 -L writable /dev/nvme0n1p2 ...

Syncing file systems (can take a long time)
Syncing mounted partitions:
     e2label /dev/nvme0n1p2 writable
  Mounting /dev/nvme0n1p2 on /mnt/clone
  => rsync // /mnt/clone with-root-excludes ...
     fatlabel /dev/nvme0n1p1 system-boot
fatlabel: warning - lowercase labels might not work properly on some systems
  Mounting /dev/nvme0n1p1 on /mnt/clone/boot/firmware
  => rsync /boot/firmware/ /mnt/clone/boot/firmware  ...  Remounting /boot/firmware/

===============================
Done with clone to /dev/nvme0n1
   Start - 11:34:24    End - 11:37:00    Elapsed Time - 2:36

Cloned partitions are mounted on /mnt/clone for inspection or customizing.

Hit Enter when ready to unmount the /dev/nvme0n1 partitions ...
```

All looks well except for an unexpected `Remounting /boot/firmware/` after sync

### Step 2. Check destination /mnt/clone/boot/firmware
```
andrey@raspberry-2:/home/andrey$ ls -la /mnt/clone/boot/firmware
total 8
drwxr-xr-x 2 root root 4096 Jan  1  1970 .
drwxr-xr-x 4 root root 4096 Sep 20 06:48 ..
```
It's empty -> BUG

### Step 3. Check source /boot/firmware
```
andrey@raspberry-2:/home/andrey$ ls -la /boot/firmware
total 8
drwxr-xr-x 2 root root 4096 Jan  1  1970 .
drwxr-xr-x 4 root root 4096 Sep 20 06:48 ..
```
It's empty -> BUG

### Step 4. Check mounting points
```
andrey@raspberry-2:/home/andrey$ lsblk -f
NAME        FSTYPE   FSVER LABEL       UUID                                 FSAVAIL FSUSE% MOUNTPOINTS
.....
sda
├─sda1      vfat     FAT32 system-boot BB79-E21C
└─sda2      ext4     1.0   writable    0dc48d41-bb0b-4b18-a2b0-1f13395e7c62  420.1G     3% /
nvme0n1
├─nvme0n1p1 vfat     FAT32 system-boot 5BA7-397B                               511M     0% /boot/firmware
│                                                                                          /mnt/clone/boot/firmware
└─nvme0n1p2 ext4     1.0   writable    13fb864a-fbcc-40fb-a2a3-ccbc7f9c5a16  430.1G     3% /mnt/clone
```
BOTH `/boot/firmware` and `/mnt/clone/boot/firmware` are mounted to the same place: new partition -> BUG

moujikov, 2025-09-27:
What I believe is happening:

Somehow original mounting `sda1 -> /boot/firmware` gets unmounted during clone (see an unexpected `Remounting /boot/firmware/` during clone). There is even a workaround in code to cope with that:

https://github.com/geerlingguy/rpi-clone/blob/9d07a8c5bf4e11b77b696b09b29b814d3f2fa713/rpi-clone#L300-L311

The problem seems to be in this line using `mount` without specifying `--source`:

https://github.com/geerlingguy/rpi-clone/blob/9d07a8c5bf4e11b77b696b09b29b814d3f2fa713/rpi-clone#L304

If no source is specified it is looked up in `/etc/fstab`, and mine states:
```
andrey@raspberry-2:/home/andrey$ cat /etc/fstab
LABEL=writable	/	ext4	defaults	0	1
LABEL=system-boot	/boot/firmware	vfat	defaults	0	1
```

BUT at this point the newly created destination NVME partition is already labeled `system-boot` as well.
`mount` then takes a wrong, empty partition and remounts it at `/boot/firmware`.
Subsequent sync is obviously of no effect.

framps, 2025-09-27:
> There is even a workaround in code to cope with that:

I added this code. I spent a lot of time to find out why /boot/firmware is umounted on the RPi5 (the issue pops up on the RPi5 only). I didn't find the root cause and therefore added this workaround.

Just to make sure I get you right: With this workaround it happened for you /boot/firmware was mounted but not mounted to the source boot partition but the empty target partition?

moujikov, 2025-09-28:
Exactly.
`/boot/firmware` indeed gets occasionally unmounted on RPi5. And workaround wrongly re-mounts it from the newly created target partition instead of the source boot one. 

This happens because `mount` is invoked without device parameter and tries to look it up by mounting point path in `/etc/fstab`. Default `/etc/fstab` relies on labels (see mine above), and since source and target partitions are already labeled identically, wrong one may be chosen.

I presume `mount` should be invoked with device and path explicitly provided.

framps, 2025-09-28:
I'll take care of this issue.

framps, 2025-09-28:
@moujikov Quick question: Which OS do you use?

framps, 2025-09-28:
Thanks you. Would you please show your /etc/fstab contents?

framps, 2025-09-28:
Ok. That's it. 😄  Ubuntu uses labels for it's partitions. Therefor you face the issue with the workaround. 

I tested the workaround on my Raspberry running RaspbianOS Bookworm and RaspbianOS uses PARTUUIDs instead of labels. Therefore your issue doesn't pop up with RaspbianOS and other OS which don't use labels.

A temporary workaround for you will be to use PARTUUIDs instead of labels.
```
andrey@raspberry-2:/home/andrey$ cat /etc/fstab
PARTUUID=xxxxxxxx-02	/	ext4	defaults	0	1
PARTUUID=xxxxxxxx-01	/boot/firmware	vfat	defaults	0	1
```
Replace xxxxxxx with the result you get with
```
lsblk -o NAME,PARTUUID
```

moujikov, 2025-09-28:
> Ok. That's it. 😄 Ubuntu uses labels for it's partitions. Therefor you face the issue with the workaround.
Yep, exactly.  

> A temporary workaround for you will be to use PARTUUIDs instead of labels.
Thanks!
I actually just manually run `sudo mount /dev/sda1 /boot/firmware` in a separate console window while the main partition is being copied. The unmounting happens at a very early stage and I have enough time before boot partition's turn comes.

framps, 2025-09-28:
I created a fix in [my fork](https://github.com/framps/rpi-clone/tree/issue_70) and changed the version to `2.0.25-testfix_70
`. 

Major change was the removal of the mount in `rsync_file_system` and addition of the mount of the source  just before rsync_file_system is called. At this place the source partition is used for the mount and therefore has no issue if labels are used in /etc/fstab. 

`mount_partition` was updated to mount the partition only if it's not already mounted and in addition a message is written if the partition is already mounted. 

Finally `rsync_file_system` now aborts if the source is not mounted - which shouldn't occur any more.

For details see [the diff](https://github.com/framps/rpi-clone/commit/b528306ef49cf4cee427c39ebea8b1f1755f87a8).

I tested this fix on RaspianOS and Ubuntu 24.04 and had no failure so far. 

@moujikov Please verify in your environment the fix works as expected. Would be great if other folks also test the fix 😉

moujikov, 2025-09-29:
Thank you, @framps!
I confirm, the issue is fixed:

```
andrey@raspberry-2:/home/andrey$ sudo framps-rpi-clone/rpi-clone nvme0n1 -f -L writable
rpi-clone Version: 2.0.25-testfix_70

Booted disk: sda 500.1GB                   Destination disk: nvme0n1 512.1GB
---------------------------------------------------------------------------
Part               Size    FS     Label           Part   Size    FS     Label
1 /boot/firmware   512.0M  fat32  system-boot     1      512.0M  fat32  system-boot
2 root             465.3G  ext4   writable        2      476.4G  ext4   writable
---------------------------------------------------------------------------
== Initialize: IMAGE partition table - forced by option ==
1 /boot/firmware      (182.5M used)  : MKFS  SYNC to nvme0n1p1
2 root                (12.6G used)   : RESIZE  MKFS  SYNC to nvme0n1p2
---------------------------------------------------------------------------
-e clone fstab edit    : edit sda device entries to nvme0n1p.
-L writable            : volume label for root partition only.
Run setup script       : no.
Verbose mode           : no.
-----------------------:
** WARNING **          : All destination disk nvme0n1 data will be overwritten!
-----------------------:

Initialize and clone to the destination disk nvme0n1?  (yes/no): y

Initializing
  Imaging past partition 1 start.
  => dd if=/dev/sda of=/dev/nvme0n1 bs=1M count=5 ...
  Resizing destination disk last partition ...
    Resize success.
  Changing destination Disk ID ...
  => mkfs -t vfat -F 32 -n system-boot /dev/nvme0n1p1 ...
  => mkfs -t ext4 -L writable /dev/nvme0n1p2 ...

Syncing file systems (can take a long time)
Syncing mounted partitions:
  e2label /dev/nvme0n1p2 writable
  Mounting /dev/nvme0n1p2 on /mnt/clone
  => rsync // /mnt/clone with-root-excludes ...
  fatlabel /dev/nvme0n1p1 system-boot
fatlabel: warning - lowercase labels might not work properly on some systems
  Mounting /dev/sda1 on /boot/firmware
  Mounting /dev/nvme0n1p1 on /mnt/clone/boot/firmware
  => rsync /boot/firmware/ /mnt/clone/boot/firmware  ...

===============================
Done with clone to /dev/nvme0n1
   Start - 15:45:28    End - 15:48:04    Elapsed Time - 2:36

Cloned partitions are mounted on /mnt/clone for inspection or customizing.

Hit Enter when ready to unmount the /dev/nvme0n1 partitions ...
```

Mounts: 
```
andrey@raspberry-2:/home/andrey$ lsblk -f
NAME        FSTYPE   FSVER LABEL       UUID                                 FSAVAIL FSUSE% MOUNTPOINTS
.....
sda
├─sda1      vfat     FAT32 system-boot BB79-E21C                             328.1M    36% /boot/firmware
└─sda2      ext4     1.0   writable    0dc48d41-bb0b-4b18-a2b0-1f13395e7c62    421G     3% /
nvme0n1
├─nvme0n1p1 vfat     FAT32 system-boot F70E-110B                             328.1M    36% /mnt/clone/boot/firmware
└─nvme0n1p2 ext4     1.0   writable    dfdfb4a1-d9d6-44e4-9e85-ed0031b093a1  431.1G     3% /mnt/clone
```